RetroPie/RetroPie-Setup #1680: RetroAchievements halts load

fngreg7, 2016-09-09:
Hey guys.  I posted my findings to the subreddit, but was directed to xpost it here.  Original thread: https://www.reddit.com/r/RetroPie/comments/51sypg/ps1_loading_very_slowly_with_solution/

Essentially, I found that with RetroAchievements on, game loading halts if it is an unsupported system.  On PS1 when using a PBP, cheevos tries to find the ROM from all available sources before proceeding. This manifests as a five minute wait as all of the network calls fail to find a game before letting the run loop proceed.  

I was using lr-pcsx-rearmed and RetroPie v4.  I have my all/retroarch.cfg set with my cheevos credentials and enabled="true". To fix this, I overrode the psx/retroarch.cfg cheevos_enable="false" and made sure a per-core cfg was enabled.  I think that coding in a "don't check cheevos for this ROM" on emulators not supported by RetroAchievements would fix this.

On RetroAchievements, the supported list is here: http://retroachievements.org/gameList.php
For simplicity, the supported systems are: MegaDrive, GB, GBC, GBA, NES, PC Engine, and SNES.

joolswills, 2016-09-09:
Not sure why you are posting this here - did you read the text when opening an issue ?

>This issue tracker is NOT for Help / Support - Please use the RetroPie forum - https://retropie.org.uk/forum/

>Even if you think you have a bug please post to the forum first, as many "issues" can be solved with configuration changes.

>For ideas and enhancements, please also post to the forum first, to see if it is something we would consider and to see if other users would find it useful.

>Please remove this text before posting.

joolswills, 2016-09-09:
This isn't a bug in retropie - it's a configuration issue - just enable it for the cores you use it on - if you think the behaviour in retroarch is incorrect it needs to be reported to them

joolswills, 2016-09-09:
I see dankcushions directed you here - not sure why.

joolswills, 2016-09-09:
I don't think we need to change anything for this - but it can be documented on the wiki perhaps.

If we disable it by default for unsupported cores, and then in the future cheevos support is added, we will have problems then with users wondering why it doesn't work. The best fix would be to avoid the delay and this would have to be done in RetroArch - if they consider it an issue that warrants fixing https://github.com/libretro/RetroArch

closed 2016-09-09 by joolswills