RetroPie/RetroPie-Setup #1683: configscripts for Emulationstation only retains few buttons

zigurana, 2016-09-09:
Hi, I noticed that the scripts run after the controller config wizard is different for retroarch and emulationstation. 
The [configscript for emulationstation](https://github.com/RetroPie/RetroPie-Setup/blob/master/scriptmodules/supplementary/emulationstation/configscripts/emulationstation.sh) now effectively strips out all buttons except for u/d/l/r/A/B/leftshoulder/rightshoulder/start/select. In stock ES, this is no issue, as these are the only supported buttonpresses.

However, as I am trying to expand on this, adding additional functionalities with other buttons would be really nice, and this is holding me back.

Would you be willing to consider a PR to fix this and support the X, Y, and LRtriggers? For the stock ES, this would make no difference, as there is nothing 'listening' for these presses anyways.

Again, this is not a bug, but an enabler for us other hackers out there.

joolswills, 2016-09-09:
So long as it wont affect the vanilla ES I am happy for a PR to implement them.

zigurana, 2016-09-09:
Ok! I'll get cracking then!

zigurana, 2016-09-15:
Thanks for picking this up!

closed 2016-09-15 by zigurana