RetroPie/RetroPie-Setup #1684: PS3 controller disconnects once pi3 gets warm

gregthefirefighter, 2016-09-11:
When my raspberry pi gets warm. It disconnects from the PS3 controllers. If I unplug the raspberry pi 3 and let it cool down, it will reconnect and start working again.

This issue tracker is NOT for Help / Support - Please use the RetroPie forum - https://retropie.org.uk/forum/

Even if you think you have a bug please post to the forum first, as many "issues" can be solved with configuration changes.

For ideas and enhancements, please also post to the forum first, to see if it is something we would consider and to see if other users would find it useful.

Please remove this text before posting.

sigboe, 2016-09-12:
> When my raspberry pi gets warm. It disconnects from the PS3 controllers. If I unplug the raspberry pi 3 and let it cool down, it will reconnect and start working again.

For sake of troubleshooting, could you try to add heatsinks to the raspberry pi? A set of 3 is highly preferable since one of the chips getting warm is the USB chip. The set I picked up from my local tech dealer literally cost $4.

> Even if you think you have a bug please post to the forum first, as many "issues" can be solved with configuration changes.

joolswills, 2016-09-12:
>This issue tracker is NOT for Help / Support - Please use the RetroPie forum - https://retropie.org.uk/forum/

>Even if you think you have a bug please post to the forum first, as many "issues" can be solved with configuration changes.

>For ideas and enhancements, please also post to the forum first, to see if it is something we would consider and to see if other users would find it useful.

>Please remove this text before posting.

closed 2016-09-12 by joolswills

gregthefirefighter, 2016-09-12:
Ok thanks. I have 3 raspberry pi 3's and they all have heat sinks. I have never had this issue before and this is the first time I've used 4.0

joolswills, 2016-09-12:
Please use the forum - you will also find more people there to advise.