RetroPie/RetroPie-Setup #1694: code documentation

joolswills, 2016-09-18:
Document helper functions via Doxygen & https://github.com/Anvil/bash-doxygen

This is just a reminder for myself - initial docs are about 60% done

for anyone interested - output is something like http://malus.exotica.org.uk/~buzz/retropiedocs/html/helpers_8sh.html (no bug reports please, still a WIP - and draft)

zigurana, 2016-09-26:
This is great! Can we help, or do you want to create a first version yourself?
Where I work, doxygen-comments are placed just above the functions, makes it a breeze to go through the code and get an idea of what's going on. Are you going for the same (type of) implementation?

joolswills, 2016-09-26:
I don't need any help currently thanks.

Check the first link in my post above for info of the implementation.

joolswills, 2016-09-26:
here is an example using bash-doxygen - https://github.com/Anvil/bash-argsparse/blob/master/argsparse.sh

joolswills, 2016-10-01:
PR in https://github.com/RetroPie/RetroPie-Setup/pull/1705 - probably still some mistakes in the documentation, but it's basically complete for helpers.sh

commit 30ade6e3 (2016-10-01): added documentation builder module and api documentation for the main helpers.sh - closes #1694

sudo ./retropie_packages.sh docs

to build the documentation - will end up in "docs/" folder off the RetroPie-Setup folder.
closed 2016-10-01 by joolswills