RetroPie/RetroPie-Setup #1703: Scraper Bug

symbios24, 2016-09-28:
Hello, there is a problem/bug with the latest Retropie version 4.0.2 when i press the scrape now option to get some game images for example for pc-engine games, it throws me to the command line and it says that i don`t have rights/privileges to write to the scrape folder "/opt/retropie/configs/all/emulationstation/downloaded_images"

When i do this with an older version v3.8.1 everything is working fine

Can this be fixed please?

Thank you

joolswills, 2016-09-28:
>This issue tracker is NOT for Help / Support - Please use the RetroPie forum - https://retropie.org.uk/forum/

>Even if you think you have a bug please post to the forum first, as many "issues" can be solved with configuration changes.

closed 2016-09-28 by joolswills

symbios24, 2016-09-29:
hi i tried to post at forum first and i got an not enough tags error so i posted here after

joolswills, 2016-09-29:
You need to put in at least one tag when posting to the forum (eg the word scraper)

symbios24, 2016-09-29:
how can i put tags?

joolswills, 2016-09-29:
In the box that says Enter tags here, between 5 and 15 characters each

symbios24, 2016-10-01:
impossible i cannot do it, i tried so many times,anyway thanks