RetroPie/RetroPie-Setup #1707: Pokemini emulator

ShunLe, 2016-10-04:
Sorry, i'm french user of retropie and my english is bad. 
I run the pokemini emulator of justburn in an other raspberry os. If is /make with sdl2. The emulator run in rpi2. 
Would you start a config for run with retropie system ?

joolswills, 2016-10-05:
I will leave this open - but please note the text you removed when posting

>For ideas and enhancements, please also post to the forum first, to see if it is something we would consider and to see if other users would find it useful.

I prefer stuff posted to the forum first (due to the number of users, it keeps the issue tracker manageable).

zerojay, 2017-01-08:
There is also now a retroarch core for Pokemini based on that same emulator. Currently does not support saving however.