RetroPie/RetroPie-Setup #1716: Vice x64: Joystick Binding Problem -> New Version of Vice

jimbo27, 2016-10-10:
With the new Version of Vice (SDL2) there was a bug which results in loosing some joystick bindings at startup of Vice.
See here:
https://retropie.org.uk/forum/topic/3972/vice-c64-emulator-now-supports-sdl2

I posted this upstream and there is already a solution for this:
https://sourceforge.net/p/vice-emu/bugs/796/

There is a new version of Vice: r31978
May I ask politely someone to compile this new version for our RetroPies? Unfortunately I am not capable of doing this.

joolswills, 2016-10-12:
I will update the module when 2.4.32 comes out.

G0nz0uk, 2016-10-13:
Hello,  yes I'm getting this issue in Retropie 4.0.3,  can't wait for this to be fixed.

G0nz0uk, 2016-10-13:
I can confirm I'm getting this too on Retropie. 

Keep up the great work. 

Thanks

joolswills, 2016-10-13:
Please use the "add your reaction" button if you want to add feedback on an issue.

joolswills, 2016-10-22:
An updated vice is now included with the fix.

closed 2016-10-22 by joolswills