RetroPie/RetroPie-Setup #1724: add libasound2-plugins:i386 to pcsx2 dependencies

Twitch0815, 2016-10-26:
Tested a few systems, with different failing rates installing pcsx2 in the Retropie X86 installation. 

sudo apt-get install libasound2-plugins:i386
is the fix for me. 

Possibly problems with 64 bit os's and 32 bit dependenceis.

Twitch0815, 2016-10-26:
http://pastebin.com/af7Er565

joolswills, 2016-10-26:
The retropie module for PCSX just installs the debian package, so im not sure it would be missing a dependency but need more details. Also please provide a log and information of what you are running on. Note: as mentioned when opening a ticket - we prefer the forum to be used first as many issues can be down to configuration etc and otherwise this issue tracker would just be full of support questions.

Twitch0815, 2016-10-26:
This is on Ubuntu 16.04.1 X64 Bit   up to date on apt-get update and upgrade

joolswills, 2016-10-26:
That is an issue with your install / dependencies. Have you got 3rd party PPAs installed or did you upgrade from an earlier version ?

Twitch0815, 2016-10-26:
No this was base install.  Following Retropie X86 installation to a T.    
The only PPA package that was installed was the Nvidia PPA to upgrade the Nvidia driver

Twitch0815, 2016-10-26:
And the behavior was consistent with a seperate machine, Macbook pro identical installation log was the same but that did not have any third party PPA's  I will produce a log from that machine.

joolswills, 2016-10-26:
aah it's because it's a i386 only package.  I will need to look into that.

joolswills, 2016-10-26:
it requires other i386 packages also, so it's possible you just had some existing dependency issue on your system that you resolved when manually installing. I'll have to check on my system.

joolswills, 2016-10-26:
I did some testing - the problem is likely due to some different paclages on your system that break the pcsx dependencies - it installs fine on a clean 16.04 - I suspect the graphics driver ppa did it. I had this ppa enabled https://launchpad.net/~oibaf/+archive/ubuntu/graphics-drivers (to fix a bug on 16.04) and it broke pcsx2 installing due to mismatch of package versions and the pcsx package dependencies . I removed it and pcsx2 installs normally.

Not something that can be fixed in RetroPie-Setup

closed 2016-10-26 by joolswills