RetroPie/RetroPie-Setup #1727: libretro n64 emulators fail to load any roms

ProxyCell, 2016-10-28:
I have tried to post this to the forums today at least twice in different browsers but I consistently get the same error stating that "category does not exist" so I am here with the issue instead:

1 - Fresh install on RPi3 with the latest RetroPie image of 4.0.2
2 - Updated system, updated RetroPie script, updated a number of RetroPie packages
3 - Began testing N64 roms and the non-libretro mupen64plus works fine but the libretro ones give the same error message:

`Parameters: 0 _SYS_ n64 /home/pi/RetroPie/roms/n64/Mario Kart 64.n64
Executing: /opt/retropie/emulators/retroarch/bin/retroarch -L /opt/retropie/libretrocores/lr-glupen64/glupen64_libretro.so --config /opt/retropie/configs/n64/retroarch.cfg "/home/pi/RetroPie/roms/n64/Mario Kart 64.n64" --appendconfig /dev/shm/retroarch.cfg
Failed to open /home/pi/RetroPie/roms/n64/Mario Kart 64.srm: No such file or directory
`

joolswills, 2016-10-28:
That issue has only been seen on the forum using safari - and will be fixed when the new version of the forum software comes out. 

Not sure what the problem is with the libretro one, but looks save game related - you should launch in verbose mode to get a better log (update retropie-setup and there is an option in the runcommand menu)

ProxyCell, 2016-10-28:
Oh that explains the forum thing, both were webkit-based

It is asking for an SRM file which does not exist but why should it? At the very bottom of this log it states it can not load the video driver and exits

`Parameters: 0 _SYS_ n64 /home/pi/RetroPie/roms/n64/Mario Kart 64.n64
Executing: /opt/retropie/emulators/retroarch/bin/retroarch -L /opt/retropie/libretrocores/lr-mupen64plus/mupen64plus_libretro.so --config /opt/retropie/configs/n64/retroarch.cfg "/home/pi/RetroPie/roms/n64/Mario Kart 64.n64" --verbose --appendconfig /dev/shm/retroarch.cfg
RetroArch [INFO] :: Redirecting save file to "/home/pi/RetroPie/roms/n64/Mario Kart 64.srm".
RetroArch [INFO] :: === Build =======================================
Capabilities: NEON VFPv3 VFPv4 
Built: Oct 27 2016
RetroArch [INFO] :: Version: 1.3.6
RetroArch [INFO] :: Git: 2b543cd
RetroArch [INFO] :: =================================================
RetroArch [INFO] :: Config: loading config from: /opt/retropie/configs/n64/retroarch.cfg.
RetroArch [INFO] :: Config: appending config "/dev/shm/retroarch.cfg"
RetroArch [WARN] :: screenshot_directory is not an existing directory, ignoring ...
RetroArch [INFO] :: Resetting undo buffers.
RetroArch [INFO] :: Loading dynamic libretro core from: "/opt/retropie/libretrocores/lr-mupen64plus/mupen64plus_libretro.so"
RetroArch [INFO] :: [overrides] no core-specific overrides found at /home/pi/.config/retroarch/config/Mupen64Plus/Mupen64Plus.cfg.
RetroArch [INFO] :: [overrides] no game-specific overrides found at /home/pi/.config/retroarch/config/Mupen64Plus/Mario Kart 64.cfg.
RetroArch [INFO] :: Shaders: preset directory: /opt/retropie/emulators/retroarch/shader/presets
RetroArch [INFO] :: Shaders: no game-specific preset found at /opt/retropie/emulators/retroarch/shader/presets/Mupen64Plus/Mario Kart 64.cgp.
RetroArch [INFO] :: Shaders: no game-specific preset found at /opt/retropie/emulators/retroarch/shader/presets/Mupen64Plus/Mario Kart 64.glslp.
RetroArch [INFO] :: Shaders: no core-specific preset found at /opt/retropie/emulators/retroarch/shader/presets/Mupen64Plus/Mupen64Plus.cgp.
RetroArch [INFO] :: Shaders: no core-specific preset found at /opt/retropie/emulators/retroarch/shader/presets/Mupen64Plus/Mupen64Plus.glslp.
RetroArch [INFO] :: Environ SET_VARIABLES.
RetroArch [INFO] :: Remaps: remap directory: /opt/retropie/configs/n64/
RetroArch [INFO] :: Remaps: no game-specific remap found at /opt/retropie/configs/n64/Mupen64Plus/Mario Kart 64.rmp.
RetroArch [INFO] :: Remaps: no core-specific remap found at /opt/retropie/configs/n64/Mupen64Plus/Mupen64Plus.rmp.
RetroArch [INFO] :: Redirecting save file to "/home/pi/RetroPie/roms/n64/Mario Kart 64.srm".
RetroArch [INFO] :: Environ GET_LOG_INTERFACE.
RetroArch [INFO] :: Environ GET_PERF_INTERFACE.
RetroArch [INFO] :: Environ SET_PIXEL_FORMAT: XRGB8888.
RetroArch [INFO] :: Environ GET_RUMBLE_INTERFACE.
RetroArch [INFO] :: Loading content file: /home/pi/RetroPie/roms/n64/Mario Kart 64.n64.
RetroArch [INFO] :: Did not find a valid content patch.
RetroArch [INFO] :: CRC32: 0x32625b0 .
RetroArch [INFO] :: Environ GET_VARIABLE mupen64-screensize:
RetroArch [INFO] :: 	640x480
RetroArch [INFO] :: Environ GET_VARIABLE mupen64-audio-buffer-size:
RetroArch [INFO] :: 	2048
RetroArch [INFO] :: Environ GET_VARIABLE mupen64-gfxplugin:
RetroArch [INFO] :: 	rice
RetroArch [INFO] :: Environ GET_VARIABLE mupen64-angrylion-vioverlay:
RetroArch [INFO] :: 	disabled
RetroArch [INFO] :: Environ GET_VARIABLE mupen64-filtering:
RetroArch [INFO] :: 	automatic
RetroArch [libretro DEBUG] :: set filtering mode...
RetroArch [INFO] :: Environ GET_VARIABLE mupen64-polyoffset-factor:
RetroArch [INFO] :: 	-3.0
RetroArch [INFO] :: Environ GET_VARIABLE mupen64-polyoffset-units:
RetroArch [INFO] :: 	-3.0
RetroArch [INFO] :: Environ GET_VARIABLE mupen64-astick-deadzone:
RetroArch [INFO] :: 	15
RetroArch [INFO] :: Environ GET_VARIABLE mupen64-gfxplugin-accuracy:
RetroArch [INFO] :: 	low
RetroArch [INFO] :: Environ GET_VARIABLE mupen64-virefresh:
RetroArch [INFO] :: 	1500
RetroArch [INFO] :: Environ GET_VARIABLE mupen64-bufferswap:
RetroArch [INFO] :: 	off
RetroArch [INFO] :: Environ GET_VARIABLE mupen64-framerate:
RetroArch [INFO] :: 	original
RetroArch [INFO] :: Environ GET_VARIABLE mupen64-pak1:
RetroArch [INFO] :: 	none
RetroArch [INFO] :: Environ GET_VARIABLE mupen64-pak2:
RetroArch [INFO] :: 	none
RetroArch [INFO] :: Environ GET_VARIABLE mupen64-pak3:
RetroArch [INFO] :: 	none
RetroArch [INFO] :: Environ GET_VARIABLE mupen64-pak4:
RetroArch [INFO] :: 	none
[... 91 lines cut ...]
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: [GL]: Using 1 textures.
RetroArch [INFO] :: [GL]: Loaded 1 program(s).
RetroArch [INFO] :: [GL]: Initializing HW render (1024 x 1024).
RetroArch [INFO] :: [GL]: Max texture size: 2048 px, renderbuffer size: 2048 px.
RetroArch [INFO] :: [GL]: Supports FBO (render-to-texture).
RetroArch [ERROR] :: Cannot open video driver ... Exiting ...
RetroArch [ERROR] :: Fatal error received in: "init_video()"
`

joolswills, 2016-10-31:
I would need to see your retroarch configs also to advise.

ProxyCell, 2016-11-02:
all retroarch config:
[retroarch-all.txt](https://github.com/RetroPie/RetroPie-Setup/files/565494/retroarch-all.txt)

n64 config:
[retroarch.txt](https://github.com/RetroPie/RetroPie-Setup/files/565491/retroarch.txt)

ProxyCell, 2016-11-02:
resolved by rebuilding retroarch
thanks @joolswills - isolated issue

closed 2016-11-02 by ProxyCell

Mausy5043, 2017-12-28:
How does one rebuild retroarch?

joolswills, 2017-12-28:
@Mausy5043 please use the retropie forum - https://retropie.org.uk/forum/