RetroPie/RetroPie-Setup #1733: building ppsspp quits the retropie_setup script

ProxyCell, 2016-11-04:
the building of ppsspp is working perfectly again but at the end of it, it just quits out of the script

joolswills, 2016-11-04:
This issue tracker is NOT for Help / Support - Please use the RetroPie forum - https://retropie.org.uk/forum/

Even if you think you have a bug please post to the forum first, as many "issues" can be solved with configuration changes.

For ideas and enhancements, please also post to the forum first, to see if it is something we would consider and to see if other users would find it useful.

Please remove this text before posting.

joolswills, 2016-11-04:
Also without even a git revision or log I cannot help.

joolswills, 2016-11-04:
I tested building on my tpi3, and was unable to reproduce btw.

joolswills, 2016-11-04:
hmm. aah you mean when selecting "back"  ? could be a problem in retropie-setup - I will re-open this.

commit fcec6586 (2016-11-04): ppsspp - switch off set -e used by the ppsspp ffmpeg build script which we source - fixes #1733
closed 2016-11-04 by joolswills