RetroPie/RetroPie-Setup #1736: Virtual gamepad not working on RPi3 with RetroPie 4.0.2

leoheck, 2016-11-05:
Hi, wiki said virtual gamepad may only work "well" with RPi2. I have a RPi3 Model B, using RetroPie 4.0.2 but it is not working. I installed it from experimental menu. Probably it's an issue with ``virtualgamepad.sh`` script. Someone has tested it with RPi3? Someone has some tips to help me solve that?

joolswills, 2016-11-05:
You have posted on the forum so you should wait for a reply there first. As when opening an issue it says

>This issue tracker is NOT for Help / Support - Please use the RetroPie forum - https://retropie.org.uk/forum/

>Even if you think you have a bug please post to the forum first, as many "issues" can be solved with configuration changes.

>For ideas and enhancements, please also post to the forum first, to see if it is something we would consider and to see if other users would find it useful.

>Please remove this text before posting.

It's also quite possible it is broken as I don't think many people use it and I've not tried it for some time.

leoheck, 2016-11-05:
Yeah, I posted the issue here because I am thinking something is broken and not only misconfigured. However, I agree with you @joolswills  that I should wait for more to have some answers there but someone has posted the same issue 22 days ago which I thought it was a good time window to post it here too. But I do not want to make any troubles. Should I close this issue or can I leave it open for now?

joolswills, 2016-11-05:
You can leave it open for now until I've had a chance to take a look. Some stuff in experimental is broken btw, or may not work properly.

leoheck, 2016-11-05:
- Considering ``Virtual Gamepad`` (installed from experimental menu)
Executing it manually as its README.md states
```
sudo node main.js
```

It gives me:
```
module.js:399
    throw err;

Error: Cannot find module 'struct'
   .... 
```

- Considering ``Mobile Gamepad``  (installed from experimental menu), it shows me many ERRORs during compilation. 

I have to improve my access to the board to copy logs to post here.

leoheck, 2016-11-05:
Just installed ``Virtual Gamepad`` from its git repository and the server is working now I have the gamepad on my android. I still have to see if it works to control the retropie.

joolswills, 2016-11-05:
I haven't tested it yet, but I saw a typo which was introduced (by me) a little while ago. I have fixed it now, so if you update retropie-setup you can test it out (Although if you have installed it manually it could interfere)

leoheck, 2016-11-05:
The mannually installed version is working. I noticed that the manual installation have the ``node_modules`` folder. This folder have the missing modules. I post before.

joolswills, 2016-11-05:
if you update retropie-setup yes. The reason the node_modules were missing was because of the typo meaning some code wasn't called.

leoheck, 2016-11-05:
:), nice so I will upgrade retropie-script to what happens!

leoheck, 2016-11-05:
Still have erros, however the node server has starded.

[image]

joolswills, 2016-11-05:
Logs are stored in RetroPie-Setup/logs (gzipped)

I don't think you need to worry about those errors if the service started. However I am making some improvements to the module currently.

leoheck, 2016-11-05:
Thanks and I can help you if you wish. It is my first time using an RPi ever :). So, should I have to close the issue now or we can leave it opened until we remove all the strange errors?

joolswills, 2016-11-05:
If everything is working for you we can close it. thanks.

closed 2016-11-05 by joolswills