RetroPie/RetroPie-Setup #1741: Packaging Conflict (ubuntu 16)

xrobau, 2016-11-10:
Emulation Station requires 'libfreetype6-dev', which requires 'libpng-dev'.

'libpng-dev' is libpng1.2, which doesn't have 'png_set_longjmp_fn' which was introduced in 1.3

GLupeN64 requires this

    /GLideN64/src/GLideNHQ/TxImage.o: In function `TxImage::getPNGInfo(_IO_FILE*, png_struct_def**, png_info_def**)':
    TxImage.cpp:(.text+0xd9): undefined reference to `png_set_longjmp_fn'
    ./GLideN64/src/GLideNHQ/TxImage.o: In function `TxImage::writePNG(unsigned char*, _IO_FILE*, int, int, int, unsigned short, unsigned char*)':
    TxImage.cpp:(.text+0x4db): undefined reference to `png_set_longjmp_fn'
    collect2: error: ld returned 1 exit status
    Makefile:295: recipe for target 'glupen64_libretro.so' failed
    make: *** [glupen64_libretro.so] Error 1

joolswills, 2016-11-10:
which Ubuntu 16 ? ubuntu 16.10 is not yet supported (and there are known issues).

joolswills, 2016-11-10:
Also please give details of version of retropie-setup etc.

xrobau, 2016-11-10:
Sorry, yes, 16.04.01. 

```
root@laptop:~/RetroPie-Setup# apt-get --ignore-missing install libfreetype6-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libfreetype6-dev : Depends: libpng-dev
E: Unable to correct problems, you have held broken packages.
root@laptop:~/RetroPie-Setup# uname -a
Linux laptop 4.4.0-45-generic #66-Ubuntu SMP Wed Oct 19 14:12:37 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
root@laptop:~/RetroPie-Setup# git rev-parse  HEAD
1a6531e9c9dc99bba27b61386b0cc53f8c3e048e
root@laptop:~/RetroPie-Setup# cat /etc/lsb-release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=16.04
DISTRIB_CODENAME=xenial
DISTRIB_DESCRIPTION="Ubuntu 16.04.1 LTS"
root@laptop:~/RetroPie-Setup#
```

There's not really anything in the logs further

```
= = = = = = = = = = = = = = = = = = = = =
Getting sources for 'lr-glupen64' : N64 emu - GLupeN64 for libretro
= = = = = = = = = = = = = = = = = = = = =

git clone --depth 1 "https://github.com/loganmc10/GLupeN64.git" "/root/RetroPie-Setup/tmp/build/lr-glupen64"
Cloning into '/root/RetroPie-Setup/tmp/build/lr-glupen64'...
~/RetroPie-Setup
~/RetroPie-Setup/tmp/build/lr-glupen64 ~/RetroPie-Setup

= = = = = = = = = = = = = = = = = = = = =
Building 'lr-glupen64' : N64 emu - GLupeN64 for libretro
= = = = = = = = = = = = = = = = = = = = =

Removing additional swap
find -name "*.o" -type f -delete
find -name "*.d" -type f -delete
rm -f glupen64_libretro.so
```
... and off it goes compiling everything ...

joolswills, 2016-11-10:
I think you have a problem with the dependencies on your system - libfreetype6-dev / libpng-devshould install fine. Aptitude might give you more information.

joolswills, 2016-11-10:
ES and lr-glupen64plus build / run  fine on my ubuntu 16.10 for example.

xrobau, 2016-11-10:
Ubuntu 16.10 has libpng-dev from libpng16.   Ubuntu 16.04 has libpng-dev from libpng12.  You need to install 'libpng16-dev' to get png_set_longjmp_fn in 16.04.  But if you install that, you can't install libfreetype6-dev, which is required by emulation station.

joolswills, 2016-11-10:
I'm sure I have built it fine on Ubuntu 16.04 before. If glupen64 has changed some requirements then you should report it to them - I don't think libpng16 should be needed (builds fine on Debian jessie which only has libpng1.2)

xrobau, 2016-11-10:
I dunno what to say man. It's broken.  I'm an open source developer, so I understand you wanting to hand it upstream as a problem, but that's not really something I can do.

If I hand it upstream, they're going to want to know what, how, and where.  But, your build script seems to delete everything, and just leaves me a gzipped logfile. 

For me to take this upstream, I'm going to need to pull apart your build script to figure out what the problem is, DUPLICATE that it is a problem with them (and not something you're passing to configure), and then liaise with them to get it fixed.

I've told you what the problem is, and explained why it's a problem.  The sucky solution would be to build ES first, then remove freetype6-dev and libpng and libpng-dev and THEN install libpng16 and libpng16-dev, and then continue building.

The *root* cause is a packaging issue in 16.04 - libpng16-dev should be saying that it provides the package libpng-dev, and it's not. As 16.04 is a LTS release, it's going to be almost impossible for them to change that, and it PROBABLY is deliberate. 

So, where does this leave us?  Pretty much stuck.  I suggest you close it as 'wontfix' and mark 16.,04 as known broken, unless someone else who knows the installer wants to take the time to pull it apart and either report it upstream, or fix the issues.

joolswills, 2016-11-10:
The buildscript only deletes everything if the build succeeds - the lr-glupen64 should stop on an error and the log should contain the build errors. (and the source will be in `RetroPie-Setup/tmp/build/lr-glupen64`)

you can build from commandline with `sudo ./retropie_packages.sh lr-glupen64plus`

but ok - I'll do it for you.

xrobau, 2016-11-10:
root@laptop:~/RetroPie-Setup# ./retropie_packages.sh lr-glupen64plus
No module 'lr-glupen64plus' found for platform x86
root@laptop:~/RetroPie-Setup#

joolswills, 2016-11-10:
Sorry - without the plus. I amended my post. run `./retropie_packages.sh` without parameters for more info.

joolswills, 2016-11-10:
I'm bisecting the issue now - looks to be something introduced with some recent commits.

joolswills, 2016-11-10:
It's broken on Raspbian too now since the change - https://github.com/loganmc10/GLupeN64/issues/97

joolswills, 2016-11-10:
fixed upstream. You will need to remove the libpng16-dev and put libpng12-dev back.

xrobau, 2016-11-10:
Still building (did a complete clean checkout), but seems to be fixed.  Will update ticket when it's completed everything. Thanks!

closed 2016-11-13 by joolswills