RetroPie/RetroPie-Setup #1743: Trivial Bug: Installer is not using shallow cloning for submodules

xrobau, 2016-11-10:
While watching the installer run through, I noticed that most (all?) modules are shallow cloned.

Disregarding if that is a good idea or not (personally, I think it's a BAD idea, but that's not for this ticket), I noticed that you're not shallow cloning the SUBMODULES.

```
~/RetroPie-Setup/tmp/build/lr-ppsspp ~/RetroPie-Setup

= = = = = = = = = = = = = = = = = = = = =
Getting sources for 'lr-ppsspp' : PlayStation Portable emu - PPSSPP port for libretro
= = = = = = = = = = = = = = = = = = = = =

git clone --depth 1 "https://github.com/libretro/libretro-ppsspp.git" "/root/RetroPie-Setup/tmp/build/lr-ppsspp"
Cloning into '/root/RetroPie-Setup/tmp/build/lr-ppsspp'...
Submodule 'dx9sdk' (https://github.com/hrydgard/minidx9.git) registered for path 'dx9sdk'
Submodule 'ext/armips' (https://github.com/Kingcom/armips) registered for path 'ext/armips'
Submodule 'ffmpeg' (https://github.com/libretro/ppsspp-ffmpeg.git) registered for path 'ffmpeg'
Submodule 'lang' (https://github.com/hrydgard/ppsspp-lang.git) registered for path 'lang'
Submodule 'native' (https://github.com/libretro/ppsspp-native.git) registered for path 'native'
Submodule 'pspautotests' (https://github.com/hrydgard/pspautotests.git) registered for path 'pspautotests'
Cloning into 'dx9sdk'...
Submodule path 'dx9sdk': checked out 'ec19e643461c84dbb256f6faaaab02cba61d4edc'
Cloning into 'ext/armips'...
Submodule path 'ext/armips': checked out '8bd93be9ba715d2f5723857edacf936ae904d07c'
Cloning into 'ffmpeg'...
```

And it sits there.. This is because ffmpeg is a huge repository.

I would suggest you add `--shallow-submodules` to your git clone command, so only the required hash is retrieved.

joolswills, 2016-11-10:
This could be problematic actually - http://stackoverflow.com/questions/27188899/shallow-clone-with-submodules-in-git-how-to-use-pointed-commits-and-not-latest

closed 2016-11-10 by joolswills

xrobau, 2016-11-10:
Hmm, I think that's right. I was going to actually try it after this had finished... It may not be possible.

joolswills, 2016-11-10:
(I was using --depth since older git doesn't have --shallow-submodules) - im reverting. sorry.

xrobau, 2016-11-10:
What are the chances of you guys accepting a pull request to NOT use shallow cloning and to use persistent directories for git checkouts? (I'm on IRC at the moment, if you want to talk about it there)

joolswills, 2016-11-10:
Bearing in mind x86 is not the primary target for retropie (rpi with small sd cards are), it would have to be an optional feature etc.

joolswills, 2016-11-11:
I added a quick variable - will work for most stuff, but there may be issues

```
sudo __persistent_repos=1 ./retropie_setup.sh
sudo __persistent_repos=1 ./retropie_packages.sh
```

joolswills, 2016-11-11:
variable name likely to change in the future and may be set automatically for desktop machines, but you can give it a test now anyway.

commit d450858a (2016-11-10): use --depth 1 for submodules - fixes #1743
commit 04afde92 (2016-11-10): Revert "use --depth 1 for submodules - fixes #1743"

This reverts commit d450858a714710ae54332f083be8edd195f0292f.