RetroPie/RetroPie-Setup #1748: mupen64plus config file wrong path

uwe42, 2016-11-19:
Symptoms:
Not able to for emulator write a file to the SaveSRAMPath, therefore not able to save the game.

Error in log file (/dev/shm/runcomand.log)
`Core Warning: couldn't open eeprom file '/RetroPie/roms/n64/[game name].eep' for writing`

Reason:
Path dose not exist, it should be:
`/home/pi/RetroPie[...]` or `~/RetroPie/[...]`

Quick fix:
Create `/RetroPie/roms/n64/` and set rights, start game, show log, error gone.

Try correct fix:
edit `/opt/retropie/config/n64/mupen64puls.cfg` and change
`SaveStatePath 
SaveSRAMPath `
to `/home/pi/RetroPie[...]`
but has no effect. Tried reboot as well.

joolswills, 2016-11-19:
I broke this recently when I removed some lines from the file. Will sort.

joolswills, 2016-11-19:
done. thank.

closed 2016-11-19 by joolswills

uwe42, 2016-11-19:
thank you for the quick fix.