RetroPie/RetroPie-Setup #1755: Add updated supertux

joolswills, 2016-11-21:
This issue tracker is NOT for Help / Support - Please use the RetroPie forum - https://retropie.org.uk/forum/

Even if you think you have a bug please post to the forum first, as many "issues" can be solved with configuration changes.

For ideas and enhancements, please also post to the forum first, to see if it is something we would consider and to see if other users would find it useful.

Please remove this text before posting.

joolswills, 2016-11-21:
Github mobile sucks. Type a topic and press enter and woops.

joolswills, 2017-09-17:
Have a module for this, but performance is very bad on the RPI.

hendursaga, 2023-10-28:
@joolswills which version of RPi?

hendursaga, 2024-02-29:
And what about x86 systems? I'm guessing it's good enough.