RetroPie/RetroPie-Setup #1763: Non-ASCII names in bluetooth devices

davidgfnet, 2016-11-23:
Hello!

You've got a but in your bluez-test-device script (not sure whether that's retropie specific, raspberry or bluez) around line 57, where you print the bluetooth device alias. If the name of the device is not properly encoded python crashes with an exception "ascii codec can't decode ..."

Maybe thast should be hardened a bit more, even the chars are not properly parsed, at least it wouldn't throw an exception.

Thanks for the great work!

joolswills, 2016-11-23:
I prefer issues to be raised first on the forum as mentioned when opening an issue here.

bluez-test-device is part of bluez - and so I guess the bug is upstream - It probably lacks lacks support for utf-8 device names.

You can manually pair a device like this via `bluetoothctl` I should think.

mat128, 2017-01-22:
I just solved this in #1852 without knowing about this being an upstream bug... Will try and follow up to the bluez team when I have some time.

joolswills, 2017-01-23:
Thanks. Technically we are using some blues test tools, but hopefully they would appreciate the fix too.

closed 2024-11-02 by cmitu