RetroPie/RetroPie-Setup #1767: mode_switch breaks fbcp based setup

mrvanes, 2016-11-28:
As can be read from this discussion in the forums
https://retropie.org.uk/forum/topic/5541/fbcp-stopped-working-solved/

The (new) mode_switch() function in runcommand.sh breaks fbcp based setups.

The workaround is to put a return 1 on top of this function.

joolswills, 2016-11-28:
It's not supposed to do any mode switching of change fb unless a mode change is requested. I will look into this but it would help to know any details of your setup, and any configuration changes you have made etc. You have not changed any runcommand parameters in es_systems.cfg or asked it to switch mode for any emulator ?

joolswills, 2016-11-28:
please also post me the output of `tvservice -s` from a terminal.

joolswills, 2016-11-28:
I think I have found a problem - introduced with some recent changes which caused tvservice to be called when not needed - didn't cause an issue on my setup, but may be the cause of your problem. Please can you update the retropie-setup script from the gui and see if that resolves it.

mrvanes, 2016-11-28:
Problem solved with latest RetroPie-Setup update. Double checked that runcommand.sh mode_switch() did NOT contain remainder of 'return 1' on top.

closed 2016-11-28 by joolswills