RetroPie/RetroPie-Setup #1772: fix up hotkey hack for retroarch

joolswills, 2016-12-01:
upstream changes have broken our patch so it needs reworking.

joolswills, 2016-12-01:
the check_input_driver_block_hotkey function always seems to return true now. Twinaphex has said it is legacy code that is being phased out, so the patch may need reworking/

joolswills, 2016-12-01:
for reference here is the patch reworked - which doesn't work since `hotkey_enable` always gets set to true - if this function is going to be removed, then it will need to be done differently as mentioned.

```diff
diff --git a/input/input_driver.c b/input/input_driver.c
index 733c353..50ca001 100644
--- a/input/input_driver.c
+++ b/input/input_driver.c
@@ -117,6 +117,10 @@ static bool input_driver_nonblock_state           = false;
 static bool input_driver_flushing_input           = false;
 static bool input_driver_data_own                 = false;
 
+/* number of frames required to trigger the hotkey */
+#define HOTKEY_DELAY 5
+static unsigned hotkey_counter = 0;
+
 /**
  * input_driver_find_handle:
  * @idx                : index of driver to get handle to.
@@ -529,6 +533,7 @@ int16_t input_state(unsigned port, unsigned device,
  **/
 static bool check_input_driver_block_hotkey(void)
 {
+   bool hotkey_enable                        = false;
    settings_t *settings                      = config_get_ptr();
    const struct retro_keybind *bind          =
       &settings->input.binds[0][RARCH_ENABLE_HOTKEY];
@@ -546,7 +551,7 @@ static bool check_input_driver_block_hotkey(void)
 
    /* If we hold ENABLE_HOTKEY button, block all libretro input to allow
     * hotkeys to be bound to same keys as RetroPad. */
-  return 
+   hotkey_enable =
          (bind->key              != RETROK_UNKNOWN)
       || (bind->joykey           != NO_BTN)
       || (bind->joyaxis          != AXIS_NONE)
@@ -554,6 +559,15 @@ static bool check_input_driver_block_hotkey(void)
       || (autoconf_bind->joykey  != NO_BTN)
       || (autoconf_bind->joyaxis != AXIS_NONE);
 
+   if (hotkey_enable)
+   {
+      if (hotkey_counter < HOTKEY_DELAY)
+         hotkey_counter++;
+   }
+   else
+      hotkey_counter = 0;
+
+   return (hotkey_enable && hotkey_counter == HOTKEY_DELAY);
 }
 
 static const unsigned buttons[] = {
```

joolswills, 2016-12-02:
https://github.com/RetroPie/RetroPie-Setup/pull/1773

closed 2016-12-03 by joolswills

Marc34, 2017-01-10:
I can't install retroarch :
01_hotkey_hack.diff  failed to apply.

joolswills, 2017-01-10:
They must have changed something in the last day or so. Will rebase the patch

joolswills, 2017-01-10:
Fixed (again).

Marc34, 2017-01-11:
Thank for your help, but the problem seems to persist. I'm on ubuntu 16.10.

joolswills, 2017-01-11:
You probably didn't update RetroPie-Setup

joolswills, 2017-01-11:
actually, they are still changing the code around upstream, so it's broken again. It's pointless for me to keep adjusting this while this file is under constant editing, so I will look at it again later today.

joolswills, 2017-01-12:
There was another issue upstream, but the patch has been redone again. It should be ok now - unless upstream code is changed before you test it - cheers

joolswills, 2017-01-12:
reference - https://github.com/libretro/RetroArch/issues/4396

Marc34, 2017-01-12:
Thank, it works fine now.

jpfox, 2017-01-12:
First, thanks for your job.
I encounter a similar error with a fresh install at a different line : 
```
patching file input/input_driver.c
Hunk #2 FAILED at 908.
1 out of 2 hunks FAILED -- saving rejects to file input/input_driver.c.rej
patching file menu/widgets/menu_entry.c
Hunk #1 succeeded at 545 (offset 2 lines).
/opt/RetroPie-Setup
retroarch patch /opt/RetroPie-Setup/scriptmodules/emulators/retroarch/01_hotkey_hack.diff failed to apply
```

joolswills, 2017-01-12:
yep. upstream have changed the code again!

jpfox, 2017-01-12:
ok, I've guessed it was such a thing :-)

joolswills, 2017-01-12:
I have reworked the patch again. They just reindented some code etc this time - so quick to fix. Hopefully they will soon finish reworking the input code :)

palmerj, 2021-01-30:
Do you think this patch is needed at all under 1.9.0? I'm looking at testing an upgrade.

joolswills, 2021-01-30:
It's not. But there will be other issues with 1.9.0. we are waiting for 1.9.1 - see https://github.com/RetroPie/RetroPie-Setup/pull/3190

Would prefer our forum to be used for queries regarding this rather than comments on a 3 year old PR.

palmerj, 2021-01-30:
Ah cool. I've seen this change https://github.com/libretro/RetroArch/commit/bd17e13ef760a89a14f6c7e1d6e2b100fc8b8b42.

palmerj, 2021-01-30:
Under retroarch master do you know which patches are still required?

joolswills, 2021-01-30:
Please open a topic on the forum or see other PR. Thanks.

commit 1633b690 (2016-12-03): Merge pull request #1773 from joolswills/retroarch_hotkey_hack

retrarch - rework hotkey enable hack for upstream retroarch and re-enable patch - fixes #1772
commit 7befe65a (2016-12-03): retrarch - rework hotkey enable hack for upstream retroarch and re-enable patch - fixes #1772