RetroPie/RetroPie-Setup #1775: Enhanced runcommand.sh : each system-folder have it's own video-settings ++

tednilsen, 2016-12-03:
First off, thank you for the wonderful work you guys have done.

I've been using my own runcommand.sh that adds some enhancements, and figured I could share it and maybe you include some of my ideas:

* Store video and emu-config in each system-folder instead, that way the same emulator can have different video-modes for each system.

* Option to disable tvservice checking, usefull for embedded lcd-screens etc.

* Added more / most common render resolutions.

Please look in my repo linked below for the changes with some comments added, please feel free to make any changes you like.

Thank you :)

[https://github.com/tednilsen/RetroPie-Setup/blob/master/scriptmodules/supplementary/runcommand/runcommand.sh](url)

joolswills, 2016-12-03:
Thanks but please do PR to submit improvements (one per feature please) so changes can be reviewed.

I took a quick look through and the parts I saw with your comments would need more work to be implemented - I do plan to improve the render resolutions though (and probably make them share the same list of modes).

Not sure why tvservice needs disabling as mode switching is not done unless requested. Otherwise it's just used to get a current mode status. Perhaps you can provide more details on that.

tednilsen, 2016-12-03:
Use polling / pr? ...I don't belive in democracy in all areas in life, nope sorry. This is one of those: I know what most people don't know they are missing out on (the deniers of this method), or not fully understand - RetroPie is yours. ;) 

Make whatever work / changes you wanne. The runcommand.sh script keep changing, therefore I've keept it simple so I could easily implement the changes.

* Video and emu-config in each system-folder:
This option is a no brainer when you look at say fbalpha, - this emulator can run lots of systems but are currently limited to only one videomode. So I've either stuck with say 720p@60Hz for all the systems that I wanne emulate with fbalpha, and forced to use another emulator when wanting another videomode.
My old method was to edit the emulators.cfg for each system, and rename lr-fbalpha to lr-fbalpha-cps, but it's still a pain in the ass. Even worse when you put everything into one file for every videomode - for every rom too, that file will "come out of control" if you have lot's of roms using different modes.
The psx emulator is also another fine example, there you have ntsc 720p@60Hz and pal 720p@50Hz systems - yet again, stuck with either ntsc or pal.
All in all, it's easier to config and maintain each system - also to clean it up...

* Two reasons to have an option for tvservice to be disabled:
1: The user have made a retropie setup with various videomodes when launching the different emulators, and now the user wanne use only the default videomode (no videomode switching) for this setup. There can be several reasons for this, one is like I have: an embedded 800 x 480 @60Hz LCD screen who reports to have some videomodes that look no good on that particular screen. It's an easy fix, and works flawless (I've been using this method for over a year, when copying my livingroom-setup to my mini-screen setup).
2: It's a faster launch of the emulator when there is no tvservice running / checking before the launch.

* New render-resolutions are taken from most common and oned in various systems, but this list of mine are a bit overkill.

To make sure the user keep they'r settings, a script method could be implemented to copy over the needed settings to each systems folder.

Cheers.

joolswills, 2016-12-04:
>Use polling / pr? ...I don't belive in democracy in all areas in life, nope sorry. This is one of those: I know what most people don't know they are missing out on (the deniers of this method), or not fully understand - RetroPie is yours. ;)

PR = Pull Request - how to submit stuff back to the project.

I don't have time to manually go through to see what you changed etc. But thanks.

closed 2016-12-04 by joolswills