RetroPie/RetroPie-Setup #1784: Mirror Releases on different provider

The1andONLYdave, 2016-12-11:
Hi, 
is it possible to mirror the Releases at another page/cdn/provider?
Or even via torrent or something similar?

atm i'm getting download speeds of some KB/s after the first MBs, and if the download reaches about between 80% and over 90% it stops and displays an error (seems like a throtteling with timeout and a access denied/link expired from S3 if chrome wants to resume the download).

I'd like to vote for torrent or sourceforge :)

Greetings, 
Dave

joolswills, 2016-12-11:
>This issue tracker is NOT for Help / Support - Please use the RetroPie forum - https://retropie.org.uk/forum/

>Even if you think you have a bug please post to the forum first, as many "issues" can be solved with configuration changes.

>For ideas and enhancements, please also post to the forum first, to see if it is something we would consider and to see if other users would find it useful.

>Please remove this text before posting.

joolswills, 2016-12-11:
If you have problems with the github CDN I recommend contacting them - they use AWS so it's likely the issue is your end. I don't current have time to maintain a torrent but there are other users on the forum who provide torrents.

joolswills, 2016-12-11:
```
Saving to: ‘retropie-4.1-rpi2_rpi3.img.gz’

retropie-4.1-rpi2_rpi3.img.gz     100%[===========================================================>] 580.97M  19.5MB/s    in 31s     

2016-12-11 11:48:04 (18.7 MB/s) - ‘retropie-4.1-rpi2_rpi3.img.gz’ saved [609189397/609189397]
```

closed 2016-12-11 by joolswills