RetroPie/RetroPie-Setup #1791: installing lr-mame (experimental package) fails

4shpool, 2016-12-15:
Installing lr-mame on RetroPie 4.1 (fully updated) fails (with just in too short time for any real compilation beeing done) with something like "cannot copy *.so". According to the Setup.log (In compliance to the guide of installing the latest mame on raspian here: http://choccyhobnob.com/tutorials/compiling-mame-on-raspberry-pi/) it needs at least gcc5.0.

Here are the last lines of the setup log:

`make[1]: Entering directory '/home/pi/RetroPie-Setup/tmp/build/lr-mame/src/devices/cpu/m68000'
Linking m68kmake...
Generating M68K source files...

	Musashi v4.90 680x0, CPU32, and ColdFire emulator
	Copyright Karl Stenerud and the MAME team.

Generated 1972 opcode handlers from 523 primitives
make[1]: Leaving directory '/home/pi/RetroPie-Setup/tmp/build/lr-mame/src/devices/cpu/m68000'
GCC version 5.0 or later needed
makefile:1206: recipe for target 'build/projects/retro/mamearcade/gmake-linux/Makefile' failed
make: *** [build/projects/retro/mamearcade/gmake-linux/Makefile] Error 255
make: *** Warte auf noch nicht beendete Prozesse...
Removing additional swap
/home/pi
Could not successfully build MAME emulator - MAME (current) port for libretro (/home/pi/RetroPie-Setup/tmp/build/lr-mame/mamearcade_libretro.so not found).

Log ended at: Mi 14. Dez 21:35:39 CET 2016
Total running time: 0 hours, 4 mins, 7 secs
`

joolswills, 2016-12-15:
You didn't read the text when opening a ticket

>This issue tracker is NOT for Help / Support - Please use the RetroPie forum - https://retropie.org.uk/forum/

>Even if you think you have a bug please post to the forum first, as many "issues" can be solved with configuration changes.

Having said that, there is nothing I can do if mame now requires a newer gcc than Raspbian has available, apart from exclude lr-mame so it doesn't show up on Raspbian systems.

joolswills, 2016-12-15:
To be more verbose, I don't think it's worth the effort of including a newer gcc just for this - lr-mame is not that useful on a Raspberry Pi - the earlier mame versions work fine - is there a specific reason you need the bleeding edge mame?

4shpool, 2016-12-15:
joolswills: I've read it - and i thought about the Forum - but as this is an issue of the current retropie raspian base (jessie), and therefore something which is an "issue" not suited for the help/support forum - think of my posted issue not in the context of "I need help", but of - your current (experimental) install package will (and can)not work in the underlying OS and should either fall back to the latest (previous current) lr-mame above lr-mame2014 that will compile in jessie, or remove it from the experimental packages until gcc5 (or better 6) is available.... maybe I should have posted it as a pull request? But IMHO not in the Forum - sorry if you think otherwise about this (but I insist on having read the text ;) )

joolswills, 2016-12-15:
We prefer even bug reports to start off on the forum - as very often they are not bugs. And we prefer to make the determination.

Even legitimate reports (like this) are best started off on the forum - so they can be confirmed by others - We have many more people there, and here it's usually me that has to deal with it and I just don't have the time (RetroPie images have had >400,000 downloads, so I cannot deal with all the reports myself).

I will lock down lr-mame to an older version depending on the gcc version available.

4shpool, 2016-12-15:
Ok, that is a valid argument I wasn't prepared for - I apologize and I will keep it in mind for future reports (confirm: posting first within the Forum). Thanks!

camdogg00, 2017-01-12:
Just to note this is still an issue. As a response to why this is actually needed is that certain emulators run certain roms better. Such as Mortal Kombat. I know it runs fine on the latest version of MAME hence needing this to work. Thanks!

joolswills, 2017-01-12:
@camdogg00 yes - that's why the ticket is still open..

joolswills, 2017-01-13:
I have just disabled lr-mame on systems with gcc 4.9. There is no easy route to build an older version, as the libretro fork looks to have been redone starting in November so no older libretro version is available. You can use lr-mame2014 though.

closed 2017-01-13 by joolswills

joolswills, 2017-02-01:
Please note that a new libretro mame2016 is now available in the experimental section.