RetroPie/RetroPie-Setup #1792: allow round-robin save/load

nschloe, 2016-12-15:
Here's a suggested enhancement for RetroPie:

I'm using the save/load mechanism in RetroPie quite a bit, and mostly slot 0 only. What happens every once in a while is that I save where I didn't mean to, and since I'm only using one slot, I can't ever go back to the previous save state. The manual fix for this would of course be to use more slots than one, but perhaps this can be automated.

What I have in mind is that everytime you save, it'll save the state in the `k+1 mod num_slots`th slot. This way, the current behavior is retained (loading gives you the last saved slot by default), but you have the possibility of going back to an older save state (name `num_slots` steps back).

dankcushions, 2016-12-15:
retroarch allows you to undo last save/load in the quick menu. select + retropad X to access menu. 

although this should have been posted to the forum :)

joolswills, 2016-12-15:
ideas / Enhancements should go to the forum first as mentioned when opening an issue (which people keep ignoring). there is also a retroarch option for that afaik -  savestate_auto_index - use the forum for more help

>This issue tracker is NOT for Help / Support - Please use the RetroPie forum - https://retropie.org.uk/forum/

>Even if you think you have a bug please post to the forum first, as many "issues" can be solved with configuration changes.

>For ideas and enhancements, please also post to the forum first, to see if it is something we would consider and to see if other users would find it useful.

>Please remove this text before posting.

joolswills, 2016-12-15:
@dankcushions unfortunately people don't seem to be able to read it, apart from the last line about removing the text. That's why we are on issue 1792 - instead of 200 :-)

closed 2016-12-15 by joolswills

craigerskine, 2016-12-16:
It'd be nice if GitHub had tags for assigning issues to stuff like Enhancements, [Bugs](https://github.com/OpenEmu/OpenEmu/issues?q=is%3Aissue+is%3Aopen+label%3A%22%E2%9A%91+Bug%22), Issues, etc. Oh, wait, they do. There should be no need to reprimand every poster or have condescending placeholder text in the issue field.

Most of the time the official forums are infuriating to use. Inline images load as they come into the viewport which jumps the posts all over the place. I dread when I have to look for something in those forums.

No matter what you do, people are going to come here to post issues. You might as well embrace it since yelling at people is **not** working.

joolswills, 2016-12-16:
No one is yelling. It is working, just not as well as we would like. This place would be full of support questions, with only me and a few others to answer otherwise - users get better help on the forum - it;s for them as well as me.

The text is completely polite and not condescending at all. 

Sorry - Will not be changing the process because you don't like it.

nschloe, 2016-12-19:
If I (the OP) may chime in: I didn't feel yelled at or anything. I actually felt a little bad that I didn't read the instructions carefully enough, and I agree that I should have posted this on the forum first. I'm glad that I did get an answer nonetheless.

I'd agree with the sentiment that people will use GitHub for support requests, simply because it's so darn easy to do. If the devs are fine with closing those "issues", perhaps this isn't too bad.

craigerskine, 2016-12-22:
@nschloe, Your specific post was not what I was referring to about being yelled at... it was just sort of the last straw. There are dozens of legitimate issues/suggestions posted on here every day, but every single one of them gets the same type of response... "post in the forums" and "read the text you removed" and "we are not open to changing things to make the project/community better".

So then you head over to forums and there is a disclaimer about not positing issues if someone already had that issue.

So then you try searching and filtering. You think you found the solution and get nothing. Turns out the solution is actually in the post, but since there are so many images posted in the thread the site theme tries to load the assets and posts as they come into or scroll past the viewport, and it jumps the text all over the place every few seconds... making it near impossible and incredibly frustrating to use, read, or retain anything.

It's no wonder people use GitHub to post their issue... because that's what it was designed for and it works splendidly without all the cruft. It's also the first thing that shows up in Google search results so most issue responses pushing you off to the forums (without an actual link to the forums) is ridiculous.

I've now unstarred, unfollowed, removed my forks and improvements I've done for the RP project + all the help documents I've made, and I've moved to Lakka/Retroarch. It's much more friendly and helpful. Plus they appreciate my help and use GitHub as it was meant to be used.

joolswills, 2016-12-22:
> There are dozens of legitimate issues/suggestions posted on here every day

Except there are not. 

> So then you head over to forums and there is a disclaimer about not positing issues if someone already had that issue.

There is a sticky topic that asks people to provide information, so we can help them. To avoid questions like "it doesn't work". EG knowing what system they are running on. Seems reasonable.

Very few post on github compared to the forum - where we have had over 5000 topic in 6 months. It's the best place for users to get help.

> I've now unstarred, unfollowed, removed my forks and improvements I've done for the RP project + all the help documents I've made, and I've moved to Lakka/Retroarch. It's much more friendly and helpful. Plus they appreciate my help and use GitHub as it was meant to be used.

Interesting last paragraph - taking the football home eh? You never submitted any improvements that I remember. Seems your only contribution is to have an argument. Lakka is a great project - I'm sure you will enjoy it.