RetroPie/RetroPie-Setup #1793: Controller not working in game

JeremyFBN, 2016-12-18:
Hoping for some assistance, truly desperate at this point.  I am trying to get my controller to work in the emulators.  It works fine in the retro pie gui, but when I load an a game from any system, the d pad works, the rest don't except the Y & B buttons which also go up and down.  I have tried re mapping, didn't work, I mapped by pressing Select and Y on my keyboard (Which works perfectly), but that didn't work, I mapped all the buttons to "UP". but still only D pad and Y&B work.  I have tried using my PS4 with a usb  adapter, same problem, so I know its not the controller

Does anyone know how to fix this,  I am new to Pie and I have tried every solution I could find online.   Truly desperate, any assistance would be appreciated

HerbFargus, 2016-12-18:
You already have a post on the forum. You also failed to heed the text you deleted when you posted this. GitHub is not for support.

closed 2016-12-18 by HerbFargus

JeremyFBN, 2016-12-18:
Apologies,  I misunderstood