RetroPie/RetroPie-Setup #1796: Ran RetroPie-Setup, won't start gui/emulation station

wjrivera, 2016-12-21:
After running 
`sudo ./retropie_setup.sh`
it takes me to the "RetroPie-Setup Script" page
I hit Basic Install, then reboot.
Then it takes me back to the terminal, it won't load the Gui/Emulation Station or anything else.

RaspberryPi 2B
RetroPie 4.1

verybadsoldier, 2016-12-21:
What is the output if you try to start emulationstation manually? Preferred with parameter `--debug`.

Also make sure that you grabbed to correct installation image of RetroPie for your Pi.

michals, 2016-12-21:
same on RPi3. `emulationstation` was missing libvlc5 as found in `emulationstation 2>/tmp/stderr.log`.
`sudo apt install libvlc5` fixed that for me. I also did `sudo apt update && sudo apt upgrade`.

joolswills, 2016-12-21:
@michals you need to update the RetroPie-Setup script first which includes the additional dependencies (or use update all packages).

Please use the forum for more help.

joolswills, 2016-12-21:
>This issue tracker is NOT for Help / Support - Please use the RetroPie forum - https://retropie.org.uk/forum/

>Even if you think you have a bug please post to the forum first, as many "issues" can be solved with configuration changes.

>For ideas and enhancements, please also post to the forum first, to see if it is something we would consider and to see if other users would find it useful.

>Please remove this text before posting.

closed 2016-12-21 by joolswills

jedunbar, 2017-11-12:
Hi, all.

My Raspberry Pi 3 won't load RetroPie correctly anymore. I followed the instructions to a 'T' here: https://retropie.org.uk/docs/Manual-Installation/. What happens is this: I power on my RaspPi3, it POSTs, the RetroPie splash pops up for maybe 5 seconds, goes away, (typically here I expect to see the EmulationStation logo appear, but it doesn't) then it switches to the Raspbian GUI desktop for maybe 5 seconds, then switches back to the terminal...  I ran the RetroPie script updater. It took a while, and completed successfully, but made no difference with this issue. I am running the latest version of RetroPie as I just cloned it today (11/11/2017). 

Thoughts? I can post a video of the boot process if needed.