RetroPie/RetroPie-Setup #1798: Cannot configure 2 different controllers which have the same name

danmcgoo, 2016-12-22:
First of all, I am not sure if this issue is really related to the Retropie-Setup itself.
It is actually related to the configuration of controllers in EmulationStation and retroarch.

Currently the controllers are identified by their name (name attribute given by udev).
However, I happen to have 2 different types of gamepads which firmware is made by the same company (while the brand of the gamepads are different), so their udev name is the same ("USB, 2-axis 8-button gamepad  ").

However, the physical disposition of the buttons is different, which means that if I configure one of the gamepad, the other one will have the buttons A swapped with B and X swapped with Y.

I tried to fix the issue with the following methods:
- Remap the buttons of one controller using jscal and jscal-store. However, since EmulationStation and Retroarch relies on udev, this solution does not work (I tried to change the input driver from udev to linuxraw, but the result was the same)
- Try to change the name of the devices with a udev rule (what a fool I am). I realized this is impossible...

But I do see one solution, which would be to use the Product and Vendor ID or GUID in addition of the device name.
This solution currently works in EmulationStation. If I add the DeviceGUID in addition of the device name, I can have 2 different settings for controllers having the same name.
So it would be nice that the DeviceGUIS would not be stripped off in the es_input.cfg.

The remaining task would be to make Retroarch compatible with another way to identify the controller.

joolswills, 2016-12-22:
I can probably add the GUID to the ES config, but if If retroarch doesn't have support then it will need to be put in as a request on their bugtracker.

Did you try adding the product/vendor id to the controller configs for retroarch ?

```
input_vendor_id = XX
input_product_id = XX
```

This is a minor issue imho, as this is the first I have heard of someone with two devices, with the same name but different layouts. The easiest solution would just be to purchase another controller (vs the effort of sorting all this out).

danmcgoo, 2016-12-22:
Indeed by using the product/vendor id, it works in retroarch.

Then I guess adding those information to the generated cfg files would be welcome too ;)

By the way, I found this repository which is quite interesting and may help many people to avoid configuring their gamepad:
https://github.com/libretro/retroarch-joypad-autoconfig/tree/master/udev

I agree that this is not a common issue, but it was driving me mad because those 2 gamepads are the one I may use the most frequently (both for different reasons) and I did not want to reset the inputs everytime I switch the gamepads. It would be more annoying if I use both of them at the same time.

dankcushions, 2016-12-22:
i think this might be the same issue:
https://retropie.org.uk/forum/topic/6264/you-have-to-use-identical-controllers-for-multiplayer-games-please-prove-me-wrong

i think storing the HID/UID during the es input config would be a pretty slick fix for this as that's how retroarch does it with their auto configs. i believe there's some sort of hierarchy where it looks for an autoconfig with a matching UID & HID and if none found it uses a name match as a fallback.

there is of course the problem of controllers that are different but have the same UID/HID which i believe happens sometimes! but you can't really legislate for that :)

HerbFargus, 2016-12-22:
Yes my Dragonrise controllers were one of those, that's one of the main reasons we removed all the preconfigured controls as they were conflicting too much with certain controllers and made the controls all jammy. No simple solution

dankcushions, 2016-12-23:
i'm not saying we should use retroarch's preconfigured auto-configs, but instead that when the user saves their custom auto-config via ES's binding sequence, it includes the VID/PID. 

i guess you'd also have to change however ES retrieves controller bindings to also get them via VID/PID if previously stored.

eg, if i have 2 controllers:

1. "USB Gamepad", VID:123:PID:456
2. "USB Gamepad", VID:123:PID:789

And I plug 1 in, and configure it as usual via ES, it would save an auto-config like:
```

this way, retroarch will be able to cope with two controllers with the same USB name string, but different HID/UID.

HerbFargus, 2016-12-23:
I see what you mean. Would be an improvement.

KingVarus, 2016-12-23:
@joolswills would you mind helping me fixing my problem, i have the same problem, 2 sticks 8 buttons each, sticks work fine but if i map the buttons on the second controller in emulationstation the buttons of the first one dont work anymore...

joolswills, 2016-12-23:
@KingVarus I'm the one already trying to help you on the forum - please continue to use the forum for support. thanks.

joolswills, 2016-12-29:
GUID is now included. product/vendor id is more work

joolswills, 2016-12-29:
scrap that - needs far more though and I was rushing. missed some changes, and need to make sure everything is ok for old configs so everything is updated by GUID not device name if a GUID exists. etc. I will do a PR for it so it can have some testing.

danmcgoo, 2017-01-04:
@dankcushions you got my issue right ;)

gizmo98, 2017-03-06:
How can i access GUID in retroarch.sh?

joolswills, 2017-03-06:
`$DEVICE_GUID` (it's in the input configuration docs).

Why would you need it in retroarch.sh

joolswills, 2017-03-06:
Did you mean the input configuration retroarch.sh btw or the main module ?

gizmo98, 2017-03-06:
Is there a solution to get VID/PID of a controller? Is GUID the serial number of a controller? If true it should be possible to find VID/PID under /sys/class/input.

joolswills, 2017-03-06:
not without changes to emulation station.

GUID is SDL2 specific and is generated from a bunch of stuff.

joolswills, 2017-03-06:
what we can do which was on my list, is to get the vid/pid from system, and tell people with 2 controlels the same to make sure only one is connected at once when configuring. that way we can do it without ES changes.

joolswills, 2017-03-06:
We would need some specific config option for this, as this is a niche case (same name, with different layouts). It would be more common that someone with two controllers the same, want the same layout (and in this case only want to configure once)

joolswills, 2017-03-06:
TBH, it's pretty rare, so I'm not sure how important it is to handle this - and it can be done manually of course.

joolswills, 2017-03-06:
I think my brain got a bit confused - if they were the same joystick the VID/PID would be the same - so ignore my last comment.

gizmo98, 2017-03-06:
Ok. I also thought of a function which retrieves VID/PID with controller name. But where to you want to tell people to plug in only one controller? The message should pop up before input configuration starts. Sounds complicated without ES changes.

gizmo98, 2017-03-06:
There are for example "USB Gamepad" named controllers with different layout and VID/PID.

joolswills, 2017-03-06:
we don't - it will only matter when there are two joysticks with same name. In that case (it doesn't happen often that someone has this), they just need to do this - we can document it

joolswills, 2017-03-06:
that's the easiest workaround imho.

gizmo98, 2017-03-06:
All right!

joolswills, 2017-03-06:
Getting this VID/PID in ES would be the nicest fix - I did look into this, but there is no SDL call call to do this - it might be available in some internal struct, or we would have to get it ourselves based on the joystick ID that sdl has. It's one of those things where the work involved may not be worth it (I have only heard of two cases of people running into this issue).

jamievleeshouwer, 2017-11-28:
I have 2 xbox one s wireless controllers which I would like to use with retropie (one white and one black).  When I plug them in via USB and configure them they both have a input_device of "Microsoft X-Box One S pad" and also have identical button configurations (no problems yet). When I use them over bluetooth, the input_device name on both of them is "Xbox Wireless Controller", however they have different input button configurations (firstly both are different from usb, and secondly over bluetooth white is different from black).

For bluetooth connectivity, I have created 2 separate configs for each controller and included input_vendor_id =  "1118" for both of them and input_product_id = "765" for the white controller, whilst input_product_id = "736" for the black controller.

Individually the controllers work correctly when only 1 of the controllers are connected via bluetooth (and using its appropriate config).  However when both of them are connected via bluetooth it appears (in the yellow text... I have assigned input_device_display_name to each config to distinguish the 2) that the same config is being assigned to both controllers.  The button mapping gets really messy when either of the 2 load their incorrect configuration. I believe detection is not using the VID and PID and is still only using the device name.

Also, when I try to base config detection on only the input_vendor_id and input_product_id (no input_device), the yellow text suggests the controller is not configured.

I am running build 4.3.7 (2cb0f84) which contains the changes mentioned above.

Any help on getting both XBOX controllers working simultaneously over bluetooth would be much appreciated.

phoenixweiss, 2017-12-09:
I have the same problem.
All my 3 controllers from DataFrog has identical VID/PID, but it's totally different: NES, SNES and Genesis-like ones.
I have no idea how to separate configs on them.

Bluehorn, 2017-12-10:
Just hit the same problem: Got a package of raspberry pi 3 + retropie + nes case + nes controller + snes controller.
Both controllers report vendor id 0x0810 and device id 0xe501, but have different button assignments. So I am unable to play multi player games - either the nes pad or the snes pad is working, even if I only  need 2 buttons (as available on the NES pad).

faceoff04, 2017-12-31:
Same problem with DragonRise Generic. I try all the solutions that i see en many tutorials but doen´s work. The only way it´s a Xin-Mo XM-02 and modify the boot file. Very SAD with this problem.

faceoff04, 2018-01-02:
Solved, connecting p1 to DragonRise and p2 directly to Raspberry GPIO, works only in recalbox. Not in Retropie. Very Sad with this solution.

Jakobud, 2018-01-05:
I know this is an old thread but have you guys also considered determining which USB port a controller is plugged into to determine which player or game it should be for? Every console always had labelled ports that you plug into to determine who is player 1 and who is player 2. Maybe doing the same thing with USB ports on the RPi would be a similar type of idea.

dankcushions, 2018-01-05:
@Jakobud that's how it works I believe: 
[image]

(lowest number with connected controller is player 1, etc)

Jakobud, 2018-01-05:
Oh awesome I had no idea that was the case. So I'm a little bit confused on what the problem is then. If you can predictably determine which player a controller is by which port you plug it into, what does it matter if 2 controllers share the same name or PID, etc?

dankcushions, 2018-01-05:
@Jakobud well i guess you could rebind player 1 to different layout to player 2, but that seems a pretty inelegant solution to me. typically if you've got two controllers with the same PID/VID, but different layouts, you're going to want to use specific ones as your player 1 controller for specific games/systems, rather than for different players.

RodrigoSCordoba, 2018-10-14:
Hi to everyone!! I got here looking for a fix for the same issue with two Drafonrise controllers with the same name...
I tray to follow these steps described above but nothing worked...
UNTIL!! I got the answer, at least what worked for me, and it's the dumbest thing...
All I had to do was place the player 2 joystick in the same direction as the player 1.
When I installed the sticks, I placed them with the connectors facing each other, because it was easy to plug 'em later. So, I changed the orientation of the connector (both equally facing the same direction) and it worked like a charm!!!

Anyway, hope it help!!

P.S: I just created this account so I could answer this thread. Thanks for the help!

mking42, 2018-12-22:
> Hi to everyone!! I got here looking for a fix for the same issue with two Drafonrise controllers with the same name...
> I tray to follow these steps described above but nothing worked...
> UNTIL!! I [...]

Yeah, that is the solution if the controllers, joysticks are the same. But this is more to address the case that different controllers with different layouts have the same name. If you configure 1 then the other one is wrong, and visa versa which results in the issue where you cannot configure and use both controllers normally.

kreid90, 2020-03-28:
I have as similar issue.  Trying to get an 8 Button USB Gamepad to work simultaneously as my Tankstick (set up as a game made, not keyboard), in the Libretro Emulator for Arcade.  If the 8 button is plugged in, with Tank, only the 8 button works.   If I unplug the 8 button, reboot and only have the Tank in, the Tank will work - but never together.  Not an elegant sitatuaion to always go into the various areas to keep choosing base don ROM or emulator/collection.  I am a newbie - for the most part and have been trying to figure this out for a few weeks now. This thread is the only one I have see that is close, but my newbness keeps me from trying to determine if I am reading a solution above?  Sorry folks for the cluelessness here..   Any help would be appreciated.

a2liquidwms, 2020-07-08:
Same issue with these DragonRise NES and SNES controllers. They seem to have a monopoly on the classic usb controller manufacturing.  From what I can tell, there is not a good solution here from a system standpoint, unless there's a way to limit by button's configured.  Anyone know of any other controllers that are NOT Dragonrize?  0079:0017 (vendor: product ids)