RetroPie/RetroPie-Setup #1799: RetroPie - Mayflash Gamecube controller adaptor working in emulation station but not in ROMS

aricl, 2016-12-24:
System: Raspberry Pi 3 model b
SD: 16GB Class 10 MicroSD (Pre-installed with Raspbian OS and NOOBS installer)
Distro: RetroPie v. 4.1.5
Gamepads: Mayflash Gamecube Controller Adapter for Wii U and PC USB

This problem sounds similar to this issue that was closed on August the 9th:
https://github.com/RetroPie/RetroPie-Setup/issues/564

, with a few differences:

The issue is with a RP3 model b instead of the original RP model B+
It also concerns a more recent version of RetroPie (4.1.5 instead of 2.3)
The issue is with the Mayflash Gamecube controller adapter, not the USB SNES controllers.
Other than that, the central problem is the same: the controllers are easily configured and work fine within emulationstation, but don't work when running ROMS. I have to use the keyboard, which is not ideal.

The reason that I am posting this on here is that the original solution referred to filepaths that appear to have changed since RetroPie has been updated. Hence the solution to configuring the controllers doesn't work in my case since I can't find the correct files. I have also posted this problem on the retropie forum, but I've had no replies in 20 days, and the problem remains.

I have followed all the advice in the posted issue and updated RetroPie and its setup script. The only thing I couldn't do was configure the controllers.

joolswills, 2016-12-24:
Sometimes topics do get missed, but there are some other threads related to this that may help, and I have seen references to a driver that is needed. However, please link me to your topic, and I will reply there.

joolswills, 2016-12-24:
https://retropie.org.uk/forum/topic/4010/has-anyone-gotten-the-mayflash-gamecube-adapter-to-work-with-retropie might help.

aricl, 2016-12-25:
Thanks Jools, the link that you provided had all the advice I needed. I just followed the instructions for installing and setting up the driver that you mentioned. Thanks again, I'll close this issue.

closed 2016-12-26 by joolswills