RetroPie/RetroPie-Setup #1804: Scraper from source question

sselph, 2016-12-27:
Hi,

I know RetroPie is capable of compiling sselph/scraper from source. I was curious when this happens and what OS version, specifically trying to understand what Go versions, are compiling it. I'm having a hard time maintaining compatibility with Go 1.2 which is 3 years old. The libraries I import are moving forward.

Thanks,

joolswills, 2016-12-27:
We always build from source now (bins are used for raspbian wheezy which is no longer supported). Raspbian Jessie has golang 1.3.3 - will that be ok or still too old?

joolswills, 2016-12-27:
BTW previously we didn't include the scraper "prebuilt" in the images, and some people have been having trouble due to some github limits I think. I will include it by default in future images which should sort this.

sselph, 2016-12-27:
If you are referring to the 429 errors. It was my hash.csv file. I had a free 1GB quota on Google Cloud and that has been exceeding the free quota I guess. I signed up for a free trial that got me $300 for 60 days so will get a feel for what it costs but I may splurge and spend the ~$3 a month or so it costs to host the file.

1.3.3 is still fairly old but I'll update my build tests and see if that causes it to pass. Including a prebuilt or possibly altering the code that installs Go to pull from https://golang.org/dl/ might be a solution now that they provide arm and other platforms for download.

joolswills, 2016-12-27:
Some people saw that issue, but this was when checking out the code afair.

I would be happy to host the files for you on the retropie server btw. I can give you sftp access to an area or so.

joolswills, 2016-12-27:
yeah we could also install a newer go package if the existing system go is too old - so don't worry too much about supporting an older version if it's trouble.

sselph, 2016-12-27:
Hmm I have a paid github account seems odd it would have hit a limit but I'm surprised about how much traffic I'm seeing so who knows. I see ~400-700 unique cloners. :)

I submitted a change to update travis config to 1.3.3 from 1.2 and it seems to have fixed the tests for now https://travis-ci.org/sselph/scraper/builds/186897370

So with that fix and the free trial we have some time. Setting up Go 1.7 and I could actually set up proper vendoring. That could probably solve the issues with versioning, but I'll read up on it. I'll also monitor traffic for downloads of the files I host. I try and do the best caching I can using etags but guess it is fairly popular.

sselph, 2016-12-27:
Oh if the error was in the last day and was like:
```
src/github.com/sselph/scraper/ds/ds.go:191: undefined: http.StatusTooManyRequests
src/github.com/sselph/scraper/ds/mame.go:109: undefined: http.StatusTooManyRequests
src/github.com/sselph/scraper/ds/ovgdb.go:126: undefined: http.StatusTooManyRequests
```

That was my bad. I referenced a constant that I guess was newer that 1.3.3. I changed it to the actual number `429` instead of `http.StatusTooManyRequests`

joolswills, 2016-12-27:
Aah yes. That would have been it.

joolswills, 2016-12-29:
I will close this - please re-open or create a new issue if you need any more info.

BTW I noticed your scraper supports screenscrapers.fr - but our script doesn't - looking at the code we have a boolean flag currently, so it might need a bit of refactoring to include it - is it fully working etc/ ok to add?

sselph, 2016-12-29:
It is functional but at the moment they are upgrading their server so it is down a lot. Once they get it stable, I have changes to add MAME support or it. I can send a PR here to add it to the list of options.

closed 2016-12-29 by sselph

joolswills, 2016-12-29:
great - thanks!