RetroPie/RetroPie-Setup #1811: Retropie-setup over ssh gamepad ca control (need fix)

thesweetiger, 2016-12-30:
I meet this problem today 
I setup something on retropie with retropie-setup over a ssh (I use an old crt)
if someone use a gamepad at the same time , he control the software at the screen and also the  retropie-setup open over the ssh connection 

Also if i have the retropie setup open on a ssh connection and  I open a game and need to choose the default emulator i can't control it , I can just control the ssh !!!

Thanks to finf a fix

joolswills, 2016-12-30:
Please use the forum first as mentioned when opening an issue.

This behaviour is expected- just make sure you don't have retropie-setup open when using retropie (and wanting to use the runcommand menu etc)

closed 2016-12-30 by joolswills

thesweetiger, 2016-12-31:
Unfortunately it's a bug and not necessary to talk on a forum

joolswills, 2016-12-31:
Because that's what we ask when opening an issue

>This issue tracker is NOT for Help / Support - Please use the RetroPie forum - https://retropie.org.uk/forum/

>Even if you think you have a bug please post to the forum first, as many "issues" can be solved with configuration changes.

joolswills, 2016-12-31:
It is technically a bug, but it' sort of by design since the ssh retropie-setup will be listening for joystick control and only a single joy2key is allowed to be run (so you dont end up controlling multiple retropie-setups - I will however exclude the joystick control when running via ssh which may help).

The other advantage to using the forum first is often there are other people with the issue so it can be confirmed, and we get a good idea of the set-ups involved. You provided very little information with your report (no idea about retropie version, hardware, etc)