RetroPie/RetroPie-Setup #1818: PS1 Multi Tap is completely broken - cannot play 3 or 4 player games

joquk, 2017-01-02:
On N64, 3 and 4 player games work perfectly with all controllers I use.

On PS1, players 3 and 4 do not work.

I enable Multitap as instructed.

The game detects players 3 and 4 as being present.

However their controllers do not function.

I have found other users with same issue, and it seems a fix was submitted but is not yet present inside Retro Pie.

When will this fix be delivered?

https://retropie.org.uk/forum/topic/763/psx-multitap-retroarch-4-players/24

joquk, 2017-01-02:
I have all latest updates installed. I have setup controllers correctly. This is a reproducible bug.

I am using PS4 controllers over Usb, PS4 controllers via official Sony wireless USB adapter and PS3 controllers over bluetooth.

dankcushions, 2017-01-02:
Please stop creating issues and instead refer to the forum. For example, to get this latest fix you should install from source, as is said several times in that link. The binaries are updated periodically. 

Even if there was a bug in the psx emlator, that is not a retropie problem, but a https://github.com/libretro/pcsx_rearmed problem.

closed 2017-01-03 by joolswills