RetroPie/RetroPie-Setup #1821: Scraper source update

sselph, 2017-01-03:
Hi,

I was looking at the code and noticed that it just does `go get` but if go already has code it won't update it. https://golang.org/cmd/go/#hdr-Download_and_install_packages_and_dependencies `go get -u` would update things. I could also be misunderstanding how it is run. I don't have an internet connection to download the latest code and test how this is working or send a PR.

https://github.com/RetroPie/RetroPie-Setup/blob/master/scriptmodules/supplementary/scraper.sh#L21

If I'm mistaken, feel free to just close this.

sselph, 2017-01-06:
Ah okay that must've been it. When I had that error in my code, people had to clear out the folder manually.

closed 2017-01-06 by sselph

joolswills, 2017-01-06:
I will add an option to retropie-setup  to clear the source folder which should help.