RetroPie/RetroPie-Setup #1827: RetroPie image is bricked

paeblits, 2017-01-07:
Twice tonight I tried to install this and it bricked my image. I now see the console but keyboard is not responsive. Can't do anything so I have to re-image.

Both times that I have tried to install this I had my USB stick configured to auto mount to the /home/pie/RetroPie/roms directory from boot, just like the Wiki says. Then I installed this experimental package and my system becomes unusable. 

Next I'm going to install this on a fresh image and I will do the auto mount config of my USB after.

paeblits, 2017-01-07:
I forgot to include the console error messages:

lvl0: Tried to use unset setting ShowHidden!
terminate called after throwing an instance of 'std::out_of_range'
what(): map::at
Aborted

joolswills, 2017-01-07:
Update Emulation Station.

Please use the forum as mentioned when opening an issue for more help.

closed 2017-01-07 by joolswills