RetroPie/RetroPie-Setup #1835: scummvm joystick movement

joolswills, 2017-01-13:
on the RPI with sdl2, the mouse pointer moves large amounts and is unusable under joystick control.

Look into a workaround against scummvm or sdl2.

Ilbraschio, 2017-01-20:
I'm facing a similar problem, using joystick control behaves like the mouse pointer location is not refreshed while moving, and gets refreshed as soon as i stop to move it.

joolswills, 2017-05-24:
Will add an optional scummvm linked against sdl1 - this should fix this, and also fix the mt-32 issue - https://retropie.org.uk/forum/topic/10477/scummvm-with-mt-32-support-enabled

closed 2017-05-25 by joolswills

arielscarpinelli, 2017-12-20:
Just to note that new scummvm 2.0 fixes the joystick issues

joolswills, 2017-12-20:
Thanks. @HerbFargus is doing some testing and we will update soon (removing sdl1 build if no longer needed)

HerbFargus, 2017-12-20:
can confirm that joystick/dpad is fixed with 2.0

commit 387dce2f (2017-05-25): added scummvm-sdl1 module - fixes #1835
 * scummvm built against sdl1 due to some issues with sdl2 built (mt-32 / joystick problems)
commit c75b6e81 (2017-05-25): Merge pull request #1987 from joolswills/scummvm_sdl1

added scummvm-sdl1 module - fixes #1835