RetroPie/RetroPie-Setup #1836: Governor : ondemand

gugueU, 2017-01-13:
Hi !

since yesterday my RP3 heat more as usually in emulationstation menu (not game).
before my governor status was by default on powersave at start up.
My Runcommand was tuned for turn the governor to ondemand  only when a play a game (go back to powersafe when I quite the game.

but today my RP3 governor is ondemand even before launching any games.
 
Temperature:   66.6 C
Voltage:       1.39 V (+0.19 overvolt)
Min speed:     600 MHz
Max speed:     1400 MHz
Current speed: 1400 MHz
Governor:      ondemand

How could I go back to this previous setting ?

gugueU, 2017-01-13:
I may be found a solution here:
http://askubuntu.com/questions/410860/how-to-permanently-set-cpu-power-management-to-the-powersave-governor

after install   : sudo apt-get install cpufrequtils
and enter this command : echo 'GOVERNOR="powersave"' | sudo tee /etc/default/cpufrequtils

on the next reboot the governor was on powersafe  👍 
and just switch to ondemand for the game

The CPU temp is 10 less degrees now

joolswills, 2017-01-13:
Glad you have it sorted. This issue tracker is not the best place for support with stuff like this, so please use the forum for further help (which is mentioned when opening an issue). Thanks.

closed 2017-01-13 by joolswills

gugueU, 2017-01-13:
ok sorry