RetroPie/RetroPie-Setup #1837: [Documentation] refactoring / updates to the arcade ROM management wiki

markwkidd, 2017-01-13:
Hi @HerbFargus and others. I'm (partly) a technical writer by trade. My usefulness as a coder is limited, but I sat down this morning with my documentation hat on and spent some time 'refactoring' the Managing ROMs page.

My focus was on helping new users grasp the essential differences between ROMs for arcade emulators versus cartridge ROMs, so the edits covered the content that someone might read during their first 60 seconds on that page.

I wanted to let you know so it didn't come as a surprise when you looked in and saw some new formatting. 

The only change which I'm not confident about in retrospect was renaming the page from **Managing ROMs** to **Managing Arcade ROMs.** I think the new title is better, but I didn't realize that github would not create an automatic redirect from the old title to the new like github does when you rename a code repository. Kind of a pain.

joolswills, 2017-01-13:
Thanks for the contributions. Regarding the page name, the github wiki is rather poor/limited unfortunately. We may have to put it back to the old name or have something at the old name as we have a bunch of references to the page from the forum - but I will leave that to @HerbFargus

HerbFargus, 2017-01-13:
I get emails any time the wiki is edited, thanks for the improvements, I have to agree with @joolswills that from our analytics that page is one of the most viewed on the wiki and the name change will break that.

I'm not really sure what the best approach is for things, I've been thinking about improving the documentation as a whole: see https://github.com/RetroPie/RetroPie-Setup/issues/1833

The one thing GitHub has left me wanting on is the documentation.

markwkidd, 2017-01-13:
At least for the moment I put in a stub page with a link to the rename: https://github.com/RetroPie/RetroPie-Setup/wiki/Managing-ROMs

It will not bother me in the least if you prefer to restore the original name though.

HerbFargus, 2017-01-13:
@markwidd I'd prefer we stick with original name for now just due to all the other sources that link to it. 

I'm hoping that with the read only version of the wiki once we get that sorted, we should be able to improve the page names a bit

markwkidd, 2017-01-13:
That works. I just renamed the page to its original title. Thanks, folks.

dankcushions, 2017-01-13:
IMO this is splitting the documentation a bit. you (or someone) added stuff in the managing roms page that is also in the MAME page (suggested emulators etc).

i think we need a 'quick start' arcade guide but i think the clrmamepro stuff should be on its own section outside of that. the proliferation of these full sets is such that there should be no real reason for anyone to be rebuilding them. i think it sends people down the wrong road of "i have romset X.Y .. how do i make it work?"

markwkidd, 2017-01-13:
I didn't add new content other than two bullets. I did cut out a few sentences in fact.  :\

There is some duplication for sure. I could look specifically for redundancy on another pass through the wiki in a couple of days.

markwkidd, 2017-01-14:
I made two minor edits to continue clarifying the first section.

Then I committed a more substantial cleanup of the Verifying and Rebuilding ROMs (clrmamepro) section bearing in mind some comments in this issue. I'd love any feedback.

The next stage of the process might be to tackle all of the details about each different emulator, its ROMs, and DATs. Maybe this content can be moved to the MAME and FBA emulator-specific wiki pages?

HerbFargus, 2017-01-14:
I'd be interested in @dankcushions feedback as well. 

Personally I think this whole section is redundant and just complicates things:

https://github.com/RetroPie/RetroPie-Setup/wiki/Managing-ROMs#list-of-emulators-and-their-respective-romsets

How I envision arcade emulators as dank mentioned earlier is to maybe have:

- Arcade Quick Start Guide (explaining romsets etc)
- Rebuilding Arcade Roms (and maybe filtering lists and building dats with romlister or equivalent)
- MAME (system page with the mame specific information)
- FBA 

Thoughts?

markwkidd, 2017-01-15:
I took a crack implementing your outline @HerbFargus . As always, I'm game to reverse any or all changes. I think that division does work a lot better than the previous one.

dankcushions, 2017-01-17:
i think it's an improvement! nice work all. i think the 'getting started' section of managing roms is mostly duplication from the MAME page and probably needs re-writing. i want to try that at some point...

markwkidd, 2017-01-19:
I have enjoyed working with the various pages over the last few days while also educating myself on the support forums with regard to how the documentation was being interpreted by people who have asked for help with arcade ROMs over the last few months.

In particular I have no firsthand of knowledge of setting up Neo Geo emulation in RetroPie so I was able to do the least with the content of that page. I think that one could use more work on the existing content by someone who knows how the steps of setting this up actually work in RetroPie. :)

https://github.com/RetroPie/RetroPie-Setup/wiki/Neo-Geo

I'll also offer two 'meta' questions about the Neo Geo documentation:
* Is it important to include instructions for GnGeo-Pi? Are there any advantages to running it over the presumably more straightforward pi-fba and lr-fbalpha? If it has advantages, I suggest those be listed. Maybe if there aren't any, it can be removed?
* Should something be said about setting up Neo Geo with lr-mame2003, or something said about why it is recommended not to use lr-mame2003 and stick to FB Alpha emulators?

dankcushions, 2017-01-19:
agreed about gngeo, but i've never used it. maybe there is a reason!

mame2003 runs some (i don't think all) neogeo games but FBA runs some megadrive/master system/etc games also. i think unless there's a reason to use it to run neogeo we shouldn't muddy the documentation, just as fba shouldn't be in megadrive/etc

advmame, mame2010/14/current/etc and mame4all/imame4all run neogeo also. it's endless :)

dankcushions, 2017-01-19:
i really don't understand/like the current mame/fba pages in the wiki now :( what is the current intention of the 'managing roms' page? it introduces some concepts, but mame page introduces others.

the advice of which mame/fba emulator to use should be on their respective pages, not the managing roms page IMO. there used to be a nice summary.

i really think the key information that is needed in MAME and FBA pages is actually much smaller than is currently there. the most simple quickstart is: transfer the entire (or individual files of an 'un-merged') ROM set for the mame/fba emulator you're using. i think we don't need explanations of parents/clones/sets/chds/samples in the mame page. it's too much.

the 'managing roms' page should be dedicated to those who want to understand why these rules are there, or how to use a smaller subset of a set. this is where you need the mame infodump.

just my opinion :)

markwkidd, 2017-01-19:
I'm ready to help continue to find the optimal arrangement, including pulling apart some of the new arrangement!

@dankcushions what do you think about the first item in [the outline posted by](https://github.com/RetroPie/RetroPie-Setup/issues/1837#issuecomment-272658101) @HerbFargus ?

As a best practice I do think there should be an article somewhere along the lines of a "Arcade Quick Start Guide (explaining romsets etc)" which is separate from the page that gets into the details of rebuilding sets with clrmamepro. 

'quick start' was the the concept for my edits to Managing ROMs, and clrmamepro became the new page. If Managing ROMs isn't the 'quick start,' do you see that in another place?

dankcushions, 2017-01-20:
i agree with herb's layout, but i think the 'quick start' should be semi-duplicated in mame and fba pages, but it should be 3 or so lines at the maximum. 

IMO the mame page would be something like:

> (recommended emulators section)

> Get a romset (not an individual ROM) of the _exact_ version for the emulator you wish to use. Place _all_ of the individual .zip files in the ROM directory specified in the table above. An incorrect version or missing files will cause games to immediately exit.

> Each emulator is tied to a specific collection of files, or "romset", and will not function with earlier/later romset versions. Further, there may be dependencies between .zip files so one ROM will not load without another present. For more information, see (LINK)

markwkidd, 2017-01-20:
Here's an aside: the way that many people use the term "ROM set" is not how mamdev uses it. I have tried to 'edit around' this issue rather than trying to hew to a specific definition. Here's theirs, for reference though:

http://docs.mamedev.org/usingmame/aboutromsets.html#what-is-a-rom-image
For arcade games, a ROM image or file is a copy of all of the data inside a given chip on the arcade motherboard. For most consoles and handhelds, the individual chips are frequently (but not always) merged into a single file. As arcade machines are much more complicated in their design, you’ll typically need the data from a number of different chips on the board. Grouping all of the files from Puckman together will get you a ROM set of Puckman.

An example ROM image would be the file pm1_prg1.6e stored in the Puckman ROM set.

markwkidd, 2017-01-20:
**Regarding the Quick Start Text:**
I took out the knife and cut, cut on the first part of the Managing ROMs section. I looked at the language you drafted @dankcushions  for the 'quick start' and I incorporated it in a few places.

What is left up there is still well above the three-line goal that dankcusions is advocating but maybe it's moving in the right direction. (I wish we could make a mini Quick Start page and embed it in the MAME and FBA pages, but that doesn't seem to be possible)

**Regarding the MAME page**
@dankcushions I now wonder if it's necessary to have the big table at the top of the screen with all the emulators, version numbers, and config paths since that information is repeated below under each emulator heading.

What if there was just a clickable unordered list of the emulators at the top, like a table of contents?

**Available Emulators**
* [mame4all](https://github.com/RetroPie/RetroPie-Setup/wiki/MAME#mame4all-pi-on-sourceforge): MAME 0.37b5 via MAME4ALL
* [lr-mame4all](https://github.com/RetroPie/RetroPie-Setup/wiki/MAME#lr-imame4all-on-github) MAME 0.37b5 via RetroArch MAME 2000
* [lr-mame2003](https://github.com/RetroPie/RetroPie-Setup/wiki/MAME#lr-mame2003-on-github) MAME 0.78 via RetroArch MAME 2003

etc.
It would definitely take up less space, at least.

HerbFargus, 2017-02-06:
So I've been thinking more about it.

What if we just called the managing roms page `Arcade`? I've just about got the new wiki sorted thanks to @joolswills help. Of course we will leave a link to the new page on the old managing roms page. This is the current state but I'm going to set the arcade pages all under their own section probably above the emulators section

retropie.org.uk/docs

looks awful familiar to libretro's new docs ;)

HerbFargus, 2017-06-04:
well I think its more or less established now. I don't really know if it is reasonable to attempt maintaining compatibility sheets with how often certain emulators and subsequent romsets are updated but if people want to try then they are welcome to. They are currently in the care of @dankcushions.

Also at some point it would be cool to have a guide on filtering romlists with something like romlister, but I imagine that the type of people who are wont to do such things probably already know how or at least know which sources to go to to do it. 

I'm closing this issue out now, thanks for all the help with it. I'm sure it will continually improve as time goes on.

closed 2017-06-04 by HerbFargus