RetroPie/RetroPie-Setup #1845: NEC PC-8801 support

nakatamaho, 2017-01-18:
I have just add a NEC PC-8801 (quasi88) support for retropie.
According to http://www.eonet.ne.jp/~showtime/quasi88/index.html, it is licensed under modified BSD license. 

Note: We removed non-free part by disabling commenting out "USE_FMGEN    = 1" in the Makefile.

nakatamaho, 2017-01-18:
[patch-quasi88.txt](https://github.com/RetroPie/RetroPie-Setup/files/713927/patch-quasi88.txt)

joolswills, 2017-01-20:
Please submit this as a PR and I will review. thanks.

nakatamaho, 2017-01-20:
Hi,
https://github.com/RetroPie/RetroPie-Setup/pull/1847
(Is it correct? I'm new to raise a PR...)

joolswills, 2017-01-21:
No, afraid not. I recommend having a read on the subject and experiment with your own repository to understand how it works.

Darknior, 2017-02-03:
Woaw so exiting new 👍 

@nakatamaho  do you plan to port too PC-98 ?
I really love this computer ... and i know some people port it perfectly on xBox 1 last year.

For linux you can use the port of Neko Project II called Xnp2. However, as of this writing there are some sound problems that need to be investigated ... they are fixed on xbox and i can have the source code if you want 😎 

http://www.nonakap.org/np2/
https://www.shrinemaiden.org/forum/index.php?topic=11117.0

Thanks

cmitu, 2019-07-11:
I think this can be closed ? Both `quasi88` and `lr-quasi88` are included now in RetroPie-Setup.

cmitu, 2021-12-27:
Closing, since the emulators are included in RetroPie-Setup now.

closed 2021-12-27 by cmitu