RetroPie/RetroPie-Setup #1853: Unknown Platform

CrescendoFang, 2017-01-23:
Unknown platform - please manually set the __platform variable to one of the following: armv7-mali generic-x11 imx6 odroid-c1 rpi1 rpi2 rpi3 rpi3-64 x86

Latest Raspbian PIXEL, with latest NEXT kernel.
Raspberry Pi 3.

Current RPie script 4.1.5.

mat128, 2017-01-23:
Can you provide complete output of /proc/cpuinfo?
According to https://github.com/RetroPie/RetroPie-Setup/blob/master/scriptmodules/system.sh#L194, you should be on revision 3.

You can blank out the serial number (normally the last line) if you want.

```
laptop:~ $ ssh pi@ip_address
pi@retropie:~ $ cat /proc/cpuinfo
```

joolswills, 2017-01-23:
The current RPI script is not 4.1.5. Please update.

joolswills, 2017-01-23:
@mat128 I don't think they have those lines as the rpi platform detection code was rewritten for 4.1.9 so they have the old code (which doesn't work with the BRANCH=next kernel).

@CrescendoFang Please note the text you removed when opening this issue

This issue tracker is NOT for Help / Support - Please use the RetroPie forum - https://retropie.org.uk/forum/

Even if you think you have a bug please post to the forum first, as many "issues" can be solved with configuration changes.

For ideas and enhancements, please also post to the forum first, to see if it is something we would consider and to see if other users would find it useful.

Please remove this text before posting.

I'm going to close this as it's already fixed.

closed 2017-01-23 by joolswills