RetroPie/RetroPie-Setup #1857: Drastic - Not exiting properly, not loading properly (not consistent)

InunoTaishou, 2017-01-24:
Horrible title for a topic but:

After loading a rom a few times in drastic sometimes drastic will not load. Using the top command shows that neither emulationstation or drastic are running. Also when closing drastic (using the "Quit" hotkey) sometimes drastic will close but emulationstation will not load back up. Also checking top shows that neither are in the list. Here's my log from running my last rom.

> Parameters: 
> Executing: pushd /opt/retropie/configs/nds/drastic; /opt/retropie/emulators/drastic/drastic "/home/pi/RetroPie/roms/nds/Pokémon SoulSilver.nds"; popd
> /opt/retropie/configs/nds/drastic ~
> Starting DraStic (version r2.5.0.3)
> 
> GLES: synchronous mode
> EGL_screen: 1920x1080
> EGL_VENDOR: Broadcom
> EGL_VERSION: 1.4
> GL|ES:
>  Broadcom
>  VideoCore IV HW
>  OpenGL ES 2.0
> OpenGL ES initialized
> Checking directory /opt/retropie/configs/nds/drastic/backup: It's there.
> Checking directory /opt/retropie/configs/nds/drastic/savestates: It's there.
> Checking directory /opt/retropie/configs/nds/drastic/config: It's there.
> Checking directory /opt/retropie/configs/nds/drastic/profiles: It's there.
> Checking directory /opt/retropie/configs/nds/drastic/unzip_cache: It's there.
> Checking directory /opt/retropie/configs/nds/drastic/system: It's there.
> Checking directory /opt/retropie/configs/nds/drastic/input_record: It's there.
> Checking directory /opt/retropie/configs/nds/drastic/cheats: It's there.
> Checking directory /opt/retropie/configs/nds/drastic/slot2: It's there.
> Checking directory /opt/retropie/configs/nds/drastic/scripts: It's there.
> Setting default configuration.
> SDL Input default: joystick name Xbox 360 Wireless Receiver
> SDL input: Setting default controls for Generic X-Box Pad
> Loading directory config file /opt/retropie/configs/nds/drastic/config/drastic.cf2
> Loading config file /opt/retropie/configs/nds/drastic/config/drastic.cfg
>   1 joystick(s) found.
> Got memory mapped buffer at 0x72f0e000 (drastic_mapped_memory.dat)
> Using memory map offset 0
> Direct mapping main RAM to 2000000
> Direct mapping main RAM to 2400000
> Direct mapping main RAM to 2800000
> Direct mapping main RAM to 2c00000
>   Initializing ARM9 bus-level memory map.
>   Initializing ARM7 bus-level memory map.
>   Initializing CP15.
>   Initializing DMA.
> ***Failed to load system file nds_bios_arm9.bin.
> Can't find Nintendo ARM9 BIOS. Trying free DraStic ARM9 BIOS.
> ***Failed to load system file nds_bios_arm7.bin.
> Can't find Nintendo ARM9 BIOS. Trying free DraStic ARM7 BIOS.
> ***Failed to load system file nds_firmware.bin.
>   video_3d render targets: 0xad50cc0 0xad74d80 0xad98e40 0xadbcf00
> ALSA output: requested rate of 44100, got 44100.
> ALSA output: requested buffer size of 8192, got 8192.
> ALSA output: requested period size of 2048, got 2048.
> ALSA output: Using synchronous engine.
> 1524.326172 cycles per output sample (0.000656 samples per cycle)
> Loaded 6722 games in database file /opt/retropie/configs/nds/drastic/game_database.xml.
> Loaded cheat directory from database file /opt/retropie/configs/nds/drastic/usrcheat.dat. Loaded 3204 games.
> Offset defines (nds_system @ 0x8405000):
> #define SPU_CHANNEL_SIZE                              0x0B8
> Processing command line options.
> Loading gamecard file /home/pi/RetroPie/roms/nds/Pokémon SoulSilver.nds.
> Attempting to load lua script /opt/retropie/configs/nds/drastic/scripts/Pokémon SoulSilver.lua
> Attempting to load lua script /opt/retropie/configs/nds/drastic/scripts/default.lua
> GLES_FX:: file not found: /opt/retropie/configs/nds/drastic/shaders/Linear.dfx
> GLES_FX:: loaded fallback shader
> FX loading error
> Initializing screen: scale 0 orientation 1 depth 32
> Loading config file /opt/retropie/configs/nds/drastic/config/drastic.cfg
> Loading config file /opt/retropie/configs/nds/drastic/config/Pokémon SoulSilver.cfg
> Config file /opt/retropie/configs/nds/drastic/config/Pokémon SoulSilver.cfg does not exist.
[... 97 lines cut ...]
> GLES_FX:: file not found: /opt/retropie/configs/nds/drastic/shaders/Linear.dfx
> GLES_FX:: loaded fallback shader
> FX loading error
> Initializing screen: scale 0 orientation 1 depth 32
> Saving state to /opt/retropie/configs/nds/drastic/savestates/_savestate_temp.dss.
> RTC: Loaded savestate time Mon Jan 23 21:41:36 2017
> Saving state to /opt/retropie/configs/nds/drastic/savestates/_savestate_temp.dss.
> RTC: Loaded savestate time Mon Jan 23 21:41:36 2017
> Can't savestate, busy saving old one, waiting..
>  Compressed savestate from 6083203 to 1682132 bytes.
> Renaming savestate file to /opt/retropie/configs/nds/drastic/savestates/Pokémon SoulSilver_0.dss.
>  Compressed savestate from 6083203 to 1681263 bytes.
> Renaming savestate file to /opt/retropie/configs/nds/drastic/savestates/Pokémon SoulSilver_0.dss.
> 0 mini hash hits out of 0 accesses (nan%)
> 0 hash accesses:
>  nan% hit in one hop
>  nan% hit in two hops
>  nan% hit in three hops
>  nan% hit in four or more hops
> Saving directory config to file named /opt/retropie/configs/nds/drastic/config/drastic.cf2
> Avg. GLES update 1.63ms
> ~

Also noticed the fastfoward hotkey does not actually work, unless I'm assuming what it's supposed to do isn't what it actually does. It should increase the speed of the game correct? If so it does not. I've tried multiple different shortcuts, on my 360 controller and keyboard, and neither work. This is probably a drastic issue and nothing to do with retropie.
Also another, not issue but request, that might not be possible with retropie. Increasing the cursor movement and removing the speed % in the top left corner. In the config you can disable the speed but it still shows 0.00% 0.00%

robertybob, 2017-01-24:
[Have you tried their forum?](http://www.drastic-ds.com/viewtopic.php?f=7&t=3472&start=30)

joolswills, 2017-01-24:
Problems should be reported to the forum first as clearly detailed when opening an issue here

>This issue tracker is NOT for Help / Support - Please use the RetroPie forum - https://retropie.org.uk/forum/

>Even if you think you have a bug please post to the forum first, as many "issues" can be solved with configuration changes.

Also this emulator is in "experimental" so expect issues. It's a preliminary release. It's also closed source, so there is nothing we can do about any problems - your only option is to report it to the authors in the link @robertybob provided.

closed 2017-01-24 by joolswills