RetroPie/RetroPie-Setup #1859: incorporate lr-mame2003 hiscore.dat by default

markwkidd, 2017-01-26:
This issue follows a forum thread on the same topic: https://retropie.org.uk/forum/topic/7627/lr-mame2003-hiscore-dat-could-now-be-added-automatically-when-core-is-installed/2

As I understand the scenario, there are two approaches available:

1. Amend RetroPie-Setup so that it copies `hiscore.dat` into the appropriate folder(s) as specified in the docs: https://github.com/RetroPie/RetroPie-Setup/wiki/lr-mame2003#cheats
2. Amend mame2003-libretro itself so that `hiscore.dat` is loaded from it's centralized/built-in location rather than needing to be copied in the first place

It seems like @dankcushions may be interested in this idea and suggested option 2:https://retropie.org.uk/forum/post/51838

Originally I was only thinking of option 1 but the second approach does sound better if I'm understanding it correctly.

joolswills, 2017-01-26:
I'm happy for either. One advantage of "copying" it to the location (if it doesn't exist), is a user would be able to manually update it which is what we do with some other things in retropie. Where would the "built in" location be ?

joolswills, 2017-01-26:
I would add, I don't know if anyone would need to update the hiscore.dat - but it's technically possible I guess (if say they had some offsets for the score data that were not included in the shipped one).

markwkidd, 2017-01-26:
@joolswills It might be nice if those kinds of corrections got submitted as updates to the 'master' `highscore.dat` so that they would percolate out to everyone but that may be a lot to ask users to figure out.

I have been thinking about that scenario in terms of the `catver.ini`. The genre categorizations could definitely be improved, but I'm not sure whether it would actually work to try to prompt users to submit github PRs with fixes on that either.

joolswills, 2017-01-26:
@markwkidd a good point - so I am happy to go with either option - @dankcushions let me know what you want to do.

joolswills, 2017-01-28:
Im going to copy it for now. If something changes we can always update the module.

joolswills, 2017-01-28:
I'll wait for https://github.com/libretro/mame2003-libretro/pull/80 though.

markwkidd, 2017-01-29:
@joolswills the PR got merged - now`hiscore.dat` has the right filename. O:-)

closed 2017-01-29 by joolswills

dankcushions, 2017-02-27:
FYI i had a look into incorporating the .dat into the binary somehow (it looks like libretro cores are only ever a single binary file(?)), and it seems possible but probably more trouble than it's worth. so on that basis i think this is the right approach :) After a few months we can strip the old stuff out of the mame2003 guidelines.

commit de59962c (2017-01-29): lr-mame2003 - copy included hiscore.dat to "$biosdir/mame2003/" - fixes #1859