RetroPie/RetroPie-Setup #1861: Plymouth entry is removed from cmdline.txt after disabling / enabling splashscreen on boot

movisman, 2017-01-27:
Hi,

If you configure the splashscreen from the setup script, and select the option to disable the splashscreen on boot, cmdline.txt gets changed and removes plymouth.enable=0 from the file. If you re-enable the splashscreen, the setting is not restored.

Cheers

joolswills, 2017-01-27:
as a workaround for now instead of "disabling" just remove the splashscreen module completely and reinstall it and it will disable plymouth again. I need to move some functions around. thanks.

movisman, 2017-01-27:
No worries, I actually stumbled across it by accident, as I selected the disable option by mistake whilst in the menu. Then I noticed console output and realised something must have changed inside the cmdline.txt. Just thought i'd give you a heads up.

Cheers for the help.

joolswills, 2017-01-27:
This should be resolved now. Note the behaviour is now:

 * only disable plymouth when screensaver module is installed
 * enable plymouth when screensaver module is removed - not when it is just disabled

commit 2dc7d6e6 (2017-01-27): splashscreen - plymouth changes - fixes #1861
 * only disable plymouth when screensaver module is installed
 * enable plymouth when screensaver module is removed - not when it is just disabled
only disable plymouth in /boot/cmd
closed 2017-01-27 by joolswills