RetroPie/RetroPie-Setup #1870: Steam Controller Driver causing stuttering across

Aryetis, 2017-02-01:
sc-xbox.py script is looping every 2 seconds when not detecting a steam controller. Causing the whole system to stutter every 2 seconds. The problem persists upon reboot and update of the system. Only way to fix it is to reinstall the "steam controller driver" package from retropie-setup.

Syslog : https://hijackedbrain.com/data/syslog
snipet from Syslog
`
Feb  1 20:21:47 retropie kernel: [  139.181329] input: Steam Controller Keyboard as /devices/virtual/input/input56
Feb  1 20:21:47 retropie /sc-xbox.py: sc-xbox.py: No SteamControler Device found
Feb  1 20:21:47 retropie /sc-xbox.py: Traceback (most recent call last):#012  File "/opt/retropie/supplementary/steamcontroller/lib/python3.4/site-packages/python_steamcontroller-1.2-py3.4-linux-armv7l.egg/steamcontroller/daemon.py", line 98, in start#012    self.run()#012  File "/opt/retropie/supplementary/steamcontroller/lib/python3.4/site-packages/python_steamcontroller-1.2-py3.4-linux-armv7l.egg/EGG-INFO/scripts/sc-xbox.py", line 70, in run#012    sc = SteamController(callback=evm.process)#012  File "/opt/retropie/supplementary/steamcontroller/lib/python3.4/site-packages/python_steamcontroller-1.2-py3.4-linux-armv7l.egg/steamcontroller/__init__.py", line 144, in __init__#012    raise ValueError('No SteamControler Device found')#012ValueError: No SteamControler Device found`

Step to reproduce : ???? Unknown yet, the problem hasn't occured yet again. I will update this thread when I find how to reproduce the bug.

Link to the related retropie forum's thread  : https://retropie.org.uk/forum/topic/7828/stuttering-issue-across-all-the-distrib

joolswills, 2017-02-01:
You have it installed without having a steam controller ? this is a third party tool - so any problem with the script itself can be reported to https://github.com/ynsta/steamcontroller

Aryetis, 2017-02-01:
I have a steam controller and it was hooked up tried both wired, and wireless mode but none worked in xbox360 mode.

I'll take a look at their github and see if this issue has been reported.

joolswills, 2017-02-01:
Thanks. just checking (sometimes we have had people install all the drivers even though they didn't have the hardware which can be problematic).

Perhaps upstream can change the way it scans to avoid the stutter. It might be worth seeing if running it at a lower priority helps (add `nice` before the python script it in /etc/rc.local`

Also adding `2>/dev/null` to the end of the commandline to disable error logging may help.

Aryetis, 2017-02-01:
Neither `nice` or `2>dev/null` is enough to prevent stutter.

We'll see how it goes upstream.

TheSIlentLink, 2017-11-11:
Thanks so much the stuttering was happening for months and couldn’t figure out why. Removing the driver stopped it and it works perfectly now!

cmitu, 2024-06-06:
Closing this, since the latest version of `steam-controller` has changed the way it runs and there's no polling involved anymore.

closed 2024-06-06 by cmitu