RetroPie/RetroPie-Setup #1871: Add Amiga emulator libretro cores

skaba, 2017-02-01:
There are multiple amiga emulators WIP ported to Libretro with different maturity levels.
- FS-UAE
- P-UAE
- uae4arm
Request to add them to experimental for now until they reach completion state.

joolswills, 2017-02-01:
So long as they are all functional that would be ok.

skaba, 2017-02-02:
I couldnt get uae4arm and fs-uae compile (I tried binaries from Gamestarter kodi plugin and got them to work at a level)

Managed to get p-uae compile from source, games work, savestates doesn't work for me maybe you can, see my pull request

a7mag3ddon, 2017-02-04:
uae4arm is already in retropie? i use it all the time works great, or is it a different version now? I am using 0.5 midwan / amiberry version.

joolswills, 2017-02-04:
the standalone uae4arm is in retropie - the wip libretro core is not yet.

midwan, 2017-02-07:
You might want to include Amiberry in the list of Amiga emulators of RetroPie. It's a version forked from uae4arm-rpi, but with several improvements and bug fixes. I just pushed out a binary release of v2.1, which was tested on Raspbian-based distros (including RetroPie): https://github.com/midwan/amiberry/releases

I'm currently working on an SDL2 version also, with the final aim of getting everything up to the WinUAE/FS-UAE standards and then get a libretro version out as well. But there's still work to be done in that part, so for now the current release is still based on SDL1.2.

I'll notify you when the SDL2 version is ready for release.

joolswills, 2017-02-07:
@midwan Will look into it - any reason why another fork rather than not working with the existing uae4arm developer ?

a7mag3ddon, 2017-02-07:
Thus new fork seems to be updated very regularly the old one hasn't been looked at in a long time even tho I and others logged verified bugs in it.

joolswills, 2017-02-07:
@a7mag3ddon uae4arm is still developed - last commit was only a month ago. You were asked for more information on at least one of your tickets which you didn't provide btw.

My question was for midwan though as it's a shame to fragment the already very fragmented uae ports further. But I understand in some cases this is unavoidable.

midwan, 2017-02-07:
@joolswills 
I did try to reach out to Chips from the very beginning, several times in fact. Unfortunately he wasn't interested in merging in all the updates I have been adding to his version. He has selectively merged in some parts (like the Shutdown button) but not most of them.

Besides that, I wanted to move the emulator forward and bring it up to the WinUAE 3.4.0 standards. That needs a big rewrite (which is currently underway) and thus my version will end up being quite different from Chip's anyway.

joolswills, 2017-02-07:
@midwan thanks for the clarification - anything I should be aware of different from uae4arm (eg directory structure etc), when integrating it into retropie ?

midwan, 2017-02-07:
@joolswills 
It's identical to the current uae4arm-rpi version in that respect, at least for the current version.

Future versions might have some changes regarding configuration files, but I'll try to make it backwards compatible as much as possible.

a7mag3ddon, 2017-02-07:
I didn't provide info because as you saw others did it before me ith enough for the testing to be confirmed. But that was more than 6 months ago?

midwan, 2017-02-07:
@joolswills 
One detail that might be helpful, if you provide a default config file: We've added options to use a game controller button to Quit or Shutdown the system. It's not as good as libretro yet, but at least it's possible now. For now the list of possible buttons is hardcoded, but that will be updated in a newer version.

joolswills, 2017-02-07:
@a7mag3ddon getting offtopic - last reply on this ticket from the dev was December and no response from anyone - https://github.com/Chips-fr/uae4arm-rpi/issues/29 - it helps developers when people can provide more details.

joolswills, 2017-02-07:
@midwan thanks - will look into that. I have some uae4arm changes (launch script enhancement) that allow playing games directly from ES including unpacking zips etc - so I may finish that off first, then look at implementing your fork. thanks for the heads up / info.

a7mag3ddon, 2017-02-07:
I can see what's happened, he asked for command line info said it was all working fine for him 5 months ago I turned off notifications.  3 months ago ( so 2 months after he was told how to re create the issue) he then asked for more info... I never saw his request as I thought it was dead.

skaba, 2017-02-07:
@joolswills for launchscript enhancements please checkout gamestarterpi's launcher script. https://github.com/bite-your-idols/Gamestarter-Pi/blob/master/packages/OLE-compiling/packages/gamestarter_OLD/uae4arm/scripts/uae4arm.start may be of help.

skaba, 2017-02-07:
@joolswills Can you port your changes to my lr-puae package, currently setup depends on RetroPieAssistant to generate the .uae files.

skaba, 2017-02-19:
@joolswills I started porting your uae4arm launcher to puae libretro version. I also had to rename it due to addEmulator helper function treating lr-* packages in a special way. Please check.

https://github.com/RetroPie/RetroPie-Setup/compare/master...skaba:master

midwan, 2017-03-28:
@joolswills 
I did some tests with the latest RetroPie (v4.2) and noticed that you include a custom compiled version of SDL2 2.0.5, which is great. What's even better, is that I don't see a bug I was facing with SDL2 and keyboard input being output to the background console, when running Amiberry-SDL2. This confirms my suspicion that it is related to either a) the configure flags I used when compiling SDL2 or b) the dependencies SDL2 picked up from my Raspbian system (or any it was missing).

Might I suggest that we add Amiberry-SDL2 as an optional package? It's currently the one that gets the most development, and once it's stable enough it will replace the older SDL1.x version in the master branch. The SDL1 version will be moved to a separate branch when that happens, for historical reasons.

Besides SDL2-base it does require SDL2_image and SDL2_ttf which should be placed in the same locations as SDL2 (needs to be discoverable by "sdl2-config --cflags --libs"). With those two additional packages, I was able to compile and run Amiberry-SDL2 on the latest RetroPie without problems.

And also, if you could please share the details regarding the configure flags you used when compiling SDL2 and the dependencies you had in place for it, it would help a lot! :)

cmitu, 2023-11-24:
The libretro cores for `uae4arm` and `P-UAE` are now part of RetroPie (including Amiberry as standalone emulator), so I think think this can be closed.

closed 2023-11-24 by cmitu