RetroPie/RetroPie-Setup #1874: Wiki holds incorrect advicse for placement of PSX-Rearmed bios files

a7mag3ddon, 2017-02-04:
Hello,

I believe the wiki holds incorrect advise for the location of the psx bios files, after spending a few hours on forums and this wiki playing with caps/ no caps/ different folder locations nothing would work, it would always use the HLE bios file thats built in.

I found a folder /home/pi/bios when i put them in there the emulator no longer warns me about using aa HLE bios file and then once in the bios settings allows me to scroll through the sever bin files i have.

I am using as of 04.02.2017 the latest emulator build from source and the current retropie 4.1 and 4.1.13 of setup script.

I have lots of screenshots to back this claim up if anyone wants to check it.

Thanks

dankcushions, 2017-02-04:
There is no /home/pi/bios/ folder. there is a /home/pi/RetroPie/bios/ folder which is the correct location and what is mentioned in the psx wiki page. are you saying that's wrong? maybe screenshots will help here.

joolswills, 2017-02-04:
@a7mag3ddon You are still being helped on the forum - but you are wrong about the path - unless of course you have changed your retroarch.cfg BIOS location. I will reply on the forum with more information.

closed 2017-02-04 by joolswills

a7mag3ddon, 2017-02-04:
Sorry but i havent changed any configs its a vanilla install and yes i am still saying that my roms only work in a folder /home/pi/bios/ 

As i said, i have lots of screen shots which i can post. If i can prove its wrong ( for me an others) why did you close it!?

joolswills, 2017-02-04:
I assumed you are running the libretro version above - I see from the forum thread you are possibly not - I will continue on the forum which is the correct place for help first.

joolswills, 2017-02-04:
>This issue tracker is NOT for Help / Support - Please use the RetroPie forum - https://retropie.org.uk/forum/

>Even if you think you have a bug please post to the forum first, as many "issues" can be solved with configuration changes.

And the forum thread is ongoing - so no need to open an issue here

a7mag3ddon, 2017-02-04:
yes i know that, im not here for help, i got it working as ive stated therfore i opened an issue ??

joolswills, 2017-02-04:
Because I would like it confirmed first on the forum and I have offered to look into it there.

a7mag3ddon, 2017-02-04:
Its been confirmed, however i think he is trying to make the emulator look for same bios folder as the other emulator, at the moment its still looking in /home/pi/bios for them

joolswills, 2017-02-04:
It is me you are talking to on the forum :)

a7mag3ddon, 2017-02-04:
ohh haha sorrry :-P