RetroPie/RetroPie-Setup #1876: enable cheats by default in lr-mame2003

markwkidd, 2017-02-05:
Similarly to the addition of `catver.ini` and `hiscore.dat`, the `cheat.dat` is now included in the MAME 2003 repository: https://github.com/libretro/mame2003-libretro/tree/master/metadata

I would like to suggest that this is an opportunity for RetroPie-Setup to enable cheats by default rather than being initiated by the user per this documentation: https://github.com/RetroPie/RetroPie-Setup/wiki/lr-mame2003#cheats

@joolswills this time I think I at least spelled `cheat.dat` correctly :D
@dankcushions I'm also tagging you because of your work on and interest in lr-mame2003

joolswills, 2017-02-05:
I'm happy to copy it across, but do we want it enabled by default ? Cheating at arcade games! It should be made as difficult as possible to do! ;-)

markwkidd, 2017-02-05:
LOL that is a good point. I like the idea of 'full functionality' but there may be a point at which it can become social engineering. On the other hand, 'unlimited free play' is kind of cheating to begin with.

dankcushions, 2017-02-27:
I'm indifferent about enabling cheats. as far as i know even if you enable it in the core options, you still have to go into the tab menu and turn each one on, etc. i've never used them outside of testing... honest :)

one thing we should do is copy cheat.dat from /metadata to /BIOS/mame2003, just like https://github.com/RetroPie/RetroPie-Setup/commit/de59962c2ef7db8b223e7c264244bc27ba2b0147#diff-6ac5e819ea2454f8d1e3a8d7bb41dd9a - this just saves a step in the guidelines :) agreed?

kaz321123, 2017-03-04:
The issue now is, I don't know how to enable cheats in mame2003 on NES-mini.  I have the cheat.dat in the right directory.  How do I do this?

dankcushions, 2017-03-04:
@kaz321123 please use the forum for support. https://retropie.org.uk/forum/category/8/help-and-support

joolswills, 2017-04-04:
cheats is copied now, i'm happy to leave it up to people to enable in core options themselves.

closed 2017-04-04 by joolswills