RetroPie/RetroPie-Setup #1885: SheepShaver in Retropie

eddymarley, 2017-02-10:
Hello,
I would like to know if you already have an initiation to add SheepShaver in Retropie, I am very interested in collaborating or starting the compilations.

I thank you

joolswills, 2017-02-10:
Well, it would require the desktop etc, but it should work, but I don't think many people would use it. Please note we prefer ideas to be first discussed on the forum to get user feedback/thoughts as detailed when opening an issue.

>This issue tracker is NOT for Help / Support - Please use the RetroPie forum - https://retropie.org.uk/forum/

>Even if you think you have a bug please post to the forum first, as many "issues" can be solved with configuration changes.

>**For ideas and enhancements, please also post to the forum first, to see if it is something we would consider and to see if other users would find it useful.**

eddymarley, 2017-02-10:
Ok, next time I'll post to forum thanks.