RetroPie/RetroPie-Setup #1892: Macintosh (minivmac) system considerations

HerbFargus, 2017-02-18:
carrying on from this post here:

https://retropie.org.uk/forum/topic/8128/minivmac-better-integration-in-emulationstation

I'm thinking I agree with the idea that we should keep both macintosh emulators under the same macintosh platform rather than splitting it into macintosh/macintoshplus

The other thing I wasn't sure about is whether or not minivmac should have a start script (or why basilisk has one as well?)

The reason I didn't put a start script in with minivmac is that I figured we could just launch any dsk image after calling the OS (which is hardcoded as System Tools.dsk and needs the proper bios or firmware or whatever it is: vMac.ROM) Is there any benefit to having a start script like we do with basilisk?

I am open to ideas or other considerations as to why they shouldn't be both included under the same macintosh platform?

joolswills, 2017-02-19:
basilisk has one because it's old and it was not seemed to be useful to have it so you could launch an image directly (set image name etc). It's not used much I guess either.

A start script is useful for emulators where you might want to just drop into the system without launching a game - eg to mess around in basic or so.

Happy to move it under mac - no need for macintoshplus imho.

HerbFargus, 2017-02-19:
Agreed.

Should have a launch script 

Should be under the macintosh platform.

closed 2017-04-04 by joolswills