RetroPie/RetroPie-Setup #1893: Attract Mode after new compile does not start up

estefan3112, 2017-02-18:
Error message: error while loading shared libraries: libsfml-graphics.s.o.2.3: cannot open shared object file: No such file or directory
Is there anything else I have to update for attract mode to work?
By the way: Attract Mode rules :-).

joolswills, 2017-02-18:
I'm assuming this is on a Raspberry Pi ? RetroPie v4.1 ? or ?

joolswills, 2017-02-18:
I reproduced it on a rpi - was due to a recent change.

closed 2017-02-18 by joolswills

estefan3112, 2017-02-18:
Retropie 4.1 with latest setup script, Experimental Package. Cool, I will try it tomorrow!!

estefan3112, 2017-02-19:
I can confirm that Attract Mode update from within Retropie works again, many thanks!

gpr17, 2017-10-07:
I am having a similar error, but with libsfml-graphics.so.2.4 is there an issue?

commit 22ba501c (2017-02-18): attractmode - install sfml as part of install - fixes #1893