RetroPie/RetroPie-Setup #1905: Cannot download raspberry pi 2/3 SD image

16bitretrokid, 2017-03-01:
This issue tracker is NOT for Help / Support - Please use the RetroPie forum - https://retropie.org.uk/forum/

Even if you think you have a bug please post to the forum first, as many "issues" can be solved with configuration changes.

For ideas and enhancements, please also post to the forum first, to see if it is something we would consider and to see if other users would find it useful.

Please remove this text before posting.

16bitretrokid, 2017-03-01:
Hi, Is there a reason why I cannot download Raspberry Pi 2/3 SD image. it fails and says "Forbidden"?

Thanks.

joolswills, 2017-03-03:
>This issue tracker is NOT for Help / Support - Please use the RetroPie forum - https://retropie.org.uk/forum/

Please use the forum next time as directed (if you had, you would have seen information about this)

https://www.theregister.co.uk/2017/03/01/aws_s3_outage/

closed 2017-03-03 by joolswills