RetroPie/RetroPie-Setup #1918: Large-Text themes for SDTV?

aubade, 2017-03-12:
It'd be really great if we could have some themes with nice heavy-weight bold text in >12px sizes for use on SDTV (especially when using 240p/progressive scan). Some of the current themes are readable to my eyes, but even on the best ones the fonts have one-pixel-wide strokes which are hard to read on CRT TVs. And I'd really like to allow my bifocal-needing father to play from a reasonable distance

robertybob, 2017-03-13:
Make a post on the RetroPie forum and see whether someone would be prepared to volunteer their time to create such a theme.

dankcushions, 2017-03-13:
there are some available. eg https://retropie.org.uk/forum/topic/2700/low-resolution-theme

it would be good to add the ones here to git/the manager, but i don't know how 'mature' they are. perhaps ask on that thread?

HerbFargus, 2017-03-13:
@dankcushions they have been added to the theme installer for quite some time now. 

OP: here is all you'll need to accomplish what you want. https://retropie.org.uk/docs/Creating-Your-Own-EmulationStation-Theme/

As mentioned when first opening a ticket these posts are best made on the forum.

closed 2017-03-13 by HerbFargus