RetroPie/RetroPie-Setup #1927: c64 emu (vice) update is missing a dependency

zoff99, 2017-03-16:
2 days ago i updated VICE c64 emulator and it did not work. when i started from commandline i saw libpulseaudio was not installed, but the emulator depended on it.
after manually installing with apt-get c64 emu works again.

joolswills, 2017-03-16:
Which system ? x86 linux ? Need more information. on the RPI it should be built without pulse.

joolswills, 2017-03-16:
the current vice binary for Raspberry Pi does not need pulseaudio. Do you mean libportaudio ? In which case this is already resolved - you probably didn't update retropie-setup.

zoff99, 2017-03-17:
yes on the raspberry pi3.
hmm. i think i did update retropi setup aswell. i will check.

joolswills, 2017-03-17:
I cannot reproduce so am closing - I suspect you didn't update the script. Also note you should have used the forum first for this - when opening an issue this text is displayed

>This issue tracker is NOT for Help / Support - Please use the RetroPie forum - https://retropie.org.uk/forum/

>Even if you think you have a bug please post to the forum first, as many "issues" can be solved with configuration changes.

closed 2017-03-17 by joolswills