RetroPie/RetroPie-Setup #1929: SSH connection enabled with raspi-config not configured properly (v4.2)

unRARed, 2017-03-20:
After enabling SSH, I was running into this:

```shell
ssh [local_ip] -l pi
> Connection closed by [local_ip]
```

I assume this is probably a permissions issue, but in my case `sudo apt-get purge openssh-server` and reinstalling ssh using `sudo tasksel` fixed the connection. Would probably also work by purging/reinstalling the openssh-server package with apt-get.

joolswills, 2017-03-21:
can't reproduce, but this is unrelated to RetroPie - RetroPie is just installed on top of Raspbian Lite.

closed 2017-03-21 by joolswills

unRARed, 2017-03-22:
Seems related. SSH works out of the box in 4.1, but not in 4.2. I read you explicitly disabled the SSH server in 4.2 and now require enabling with `raspi-config`. Doing so caused the issue above. This occurred on a very recently purchased Raspberry Pi 3.

joolswills, 2017-03-22:
You should use the forum as requested when opening a ticket. SSH is disabled by default as Raspbian has SSH disabled now too.

joolswills, 2017-03-22:
RetroPie is not an os. The images are just RetroPie installed on raspbian lite.

joolswills, 2017-03-22:
Also should reboot after enabling.

mydmitry, 2018-06-16:
I had same issue when used putty ver 0.56 to connect.
After updating to putty ver 0.7 all seems fine.
You can download it here:
[https://www.chiark.greenend.org.uk/~sgtatham/putty/latest.html
](https://www.chiark.greenend.org.uk/~sgtatham/putty/latest.html
)